ArduPilot/apm_planner #1060: save and replay logs

gpldecha, 2017-07-07:
I have started to use APM PLANNER and I was looking to record/save and replay the logs (EKF, etc...) .
This seems not to be possible ? 
Are there any plans for this feature to be added ?

Arne-W, 2017-07-12:
At the moment it is only possible to replay a tlog (the logs created by the apmplanner itself). Flashlog files (*.bin - stored by ardupilot on sdcard) cannot be replayed due to different format, but you can open them for post flight Analysis.
Tlogs are automatically created by the APM-Planner as soon as an UAV is connected. They are stored in apm-planner Folder in your home.

Arne-W, 2018-03-02:
will close cause no activity seems to be solved

closed 2018-03-02 by Arne-W