ArduPilot/apm_planner #1061: Graph Values shown outside grid

RHHoek, 2017-07-12:
With a recent AP2 build (2.0.25-rc1-31-gd9bae3ff5) (and before?) you can see values displayed outside the graph grid (see screenshot) with 'Show Values Under Mouse' enabled.
Expected behavior is that nothing is shown, like what happens when the cursor is outside the graph window.

This behavior is seen in the realtime (tlog) graph mode as in the open dataflash log mode. 

System Ubuntu 17.04 (Gnome and Unity7)
Project MESSAGE: Qt version 5.7.1
Project MESSAGE: Linux build x64_86
Project MESSAGE: Ubuntu Build
[image]

RHHoek, 2017-07-12:
I also observed that even 'Auto Scroll With Data' is not enabled, still values are display in the future.... :-) See screenshot.
[image]

Arne-W, 2017-07-16:
I can reproduce it. On my system it is "only" a cosmetic issue.
Let me know if there are other impacts for you.

RHHoek, 2017-07-18:
Perhaps your are right that this is just cosmetic but I don't understand where these values come from. If there is no data to be plotted, how is it possible that there are values show under the cursor?
What I mean is that the line on the graph is just lagging behind, or is the plotted line just shifted to the left. In that case the values shown under the cursor are synchronous with the plotted line.
(I hope you understand what I mean)

RHHoek, 2017-07-20:
Arne, Thank you for the time you spend to improve apmplanner.

If I move the tooltip from valid to invalid (in future) data I see the values changing. So, at first it seems the values under the tooltip are showing 'future' data. But by moving the tooltip the data is updated and again showing the last new value.
So it's my fault to interpret these changing values as 'future' data.

So I agree that this is just cosmetic, like de values show hidden by the axis.  
Again, thanks for your time.

Arne-W, 2017-07-22:
You are always welcome :smiley:

Arne-W, 2018-02-23:
As we could clarify everything I am closing this now.

closed 2018-02-23 by Arne-W