ArduPilot/apm_planner #1063: MAVLink packet loss with Pixhawk connected to Simple BGC32 mount

Dynode, 2017-07-14:
The Simple BGC32 board is connected to Pixhawk on Telem1/2, setup is for MAVLink protocol and mount (mnt_type Storm32). The SBGC32 board is connected with it's own USB cable, shown as a COM port on our PCs.

Issue: The Simple BGC GUI program (through it's own port) indicates MAVLink packets are lost when APM Planner2 is connected to the Pixhawk (through a 2nd USB cable / COM). It also happens when connecting through MAVProxy.

Anyone got an idea of what's going on with the packet loss?

auturgy, 2017-07-18:
Does the gimbal work when not connected to a GCS?  This issue is a little confusing, as Storm32 and SBGC are different, and use different protocols.
SBGC is not mavlink fluent, afaik.  Try using the settings for SBGC rather than Storm32, and see if it works any better.  http://ardupilot.org/copter/docs/common-simplebgc-gimbal.html#common-simplebgc-gimbal

Dynode, 2017-07-18:
Thanks for the response, I'll try using mnt_type = Alexmos serial, but my main concern is what is all the "lost traffic" the SBGC32 sees when a GCS is on the Pixhawk.

The joystick gimbal control works with Alexmos Serial API when directly connected to a PC/companion computer (not by using dronekit SW, just a program that implements the Serial API, directly connected to SBGC32 on COM).

Regarding Storm32 vs SBGC32: You're probably right, but my impression from the Storm32 forum is that their MAVLink implementation is not really 100% either.

Anyway, for MAVLink control of SBGC32, the Pixhawk MNT_TYPE param needs to be Storm32 MAVLink as that, to my knowledge, simply means "control by MAVLink" -- the param fits both gimbal controllers and is per available documentation online. You then set SERIALx_PROTOCOL = Mavlink2 and ArcuCopter Gimbal mode to MAVLink_Targeting. -- I have control the gimbal through this setup with the dronekit SW several times in a row already (joystick steering), it works when the connection (handshaking?) actually goes through those 15% of the times I start the dronekit program.

The reason I'm posting this is SBGC GUI counts always 255 "lost packets" on each update (about 1 - 2 Hz or so) when any GCS, be it APM Planner2, QGC or MAVProxy, is connected to the Pixhawk. -- I'm unable to figure out why, and what all those lost packets are. -- Why does the SBGC32 see so much traffic when a GCS is connected, and is it possible that this is why my program only works 15% of the time it starts?

I should point out that the program simply doesn't read the gimbal angles when the joystick control does *not* work -- it still connects to the Pixhawk (through MAVProxy!).

We'd like to do this by MAVLink, but if need be, we'll use the Serial API if it's responsive enough through the Pixhawk (mnt_type = Alexmos serial).

Thanks again for not just closing this issue like what happened on my Github issues for MAVProxy and QGC. I realized this probably is more a forum question, but the guys with know-how is definitely here on Github. ;)

auturgy, 2017-07-18:
I don't have a sbgc32, so it's hard to reproduce/ fault find.

Storm32 uses serial protocol 1 (mavlink1). That could well be the issue here.
I assume you're running latest firmware on the gimbal, and have followed setup to enable mavlink on it?

Regards,

James

Dynode, 2017-07-19:
Both MAVLink 1 and MAVLink 2 params works, on this setup at least (SBGC32 firmware 2.63 b0).

However; I just realized a mistake on my part -- dronekit should be run with Python 2.7, not 3. Gimbal control now works repeatedly as expected, both directly on COM to the Pixhawk, as well as through MAVProxy (which is one of the cases where "lost MAVLink packets" is indicated in SBGC GUI ).

So, things actually works. Response time lags a little when controlling through the Pixhawk, but that seems to be in line with what others describe online as well.

I can now only speculate on wtf actually happens on the wire into the gimbal controller boards when a GCS is connected to the Pixhawk -- somethings spams the board, but control works.

Perhaps this issue will help others down the road. (The RX-TX connection is on RC_ROLL & RC_YAW, meaning MAVLink happens on the RC_SERIAL pins (chose this in SBGC GUI MAVLink tab).)

Thanks again.

auturgy, 2017-07-19:
Glad it's working!

Regards,

James

closed 2017-07-20 by Dynode