ArduPilot/apm_planner #1071: Graph: does not understand/display mode.

AndKe, 2017-10-27:
Please see log: https://discuss.ardupilot.org/t/help-with-dataflash-autopsy/22430
it contains:
 2017-10-18 11:25:09.78: MODE {TimeUS : 308053047, Mode : 8, ModeNum : 8, Rsn : 0}

and is displayed as: 
[image]

Arne-W, 2017-10-28:
Hey AndKe long time no see :smile:
Nice to see that you are still using APM-Planner :+1: 
The problem is that there is only **ONE** data row with mode data (which is ok for a log). But the plotting library - which delivers the values for the mouse over - delivers no values for a graph consisting of only one point.
I made a workaround for this by adding a second point with the same value. I will let you know when it is merged.

AndKe, 2017-10-28:
:) yes, still using APM Planner, , and I see no alternatives for log analysis.
Using it as GCS (professionally) , for tuning new custom UAV's or when flying complex missions, (where I need very specific RTL and failsafe setup, sometimes changing it during mission)
There is no nearly as good alternative when it comes to AP2's "Full parameter list", which is very user friendly.

commit dd331536 (2017-11-13): GraphView: Graphs with only one point will be handled correctly now. Fixes #1071
closed 2017-11-13 by billbonney