ArduPilot/apm_planner #1073: Graph: segfault when loading .tlog

AndKe, 2017-11-04:
.tlog attached

[20171104 10:09:23.666 INFO ] - Plot Log loading took 0.012 seconds - 1751982 of 1751982 bytes used
[Thread 0x7fff6bfff700 (LWP 19205) exited]
[20171104 10:09:23.675 WARN ] - "The data was truncated!\n\r" "1 data corruptions\n"

Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007ffff35caf41 in qHash(QString const&, unsigned int) ()
   from /home/andre/Qt/5.9.1/gcc_64/lib/libQt5Core.so.5
[mav.zip](https://github.com/ArduPilot/apm_planner/files/1443082/mav.zip)

Arne-W, 2017-11-04:
Ok found the reason for the crash - there is no data in the log and an array access was unchecked and out of bounds.
But could it be possible that the log is empty?? Only one Hartbeat message without any data is in the log you attached.

AndKe, 2017-11-04:
you can dump it using mavlogdump.py , it looks just fine then..

Arne-W, 2017-11-05:
Hmm - you are right, using mavlogdump.py works somehow.
Using the C-api the parser always delivers `MAVLINK_FRAMING_INCOMPLETE` and we are waiting for a `MAVLINK_FRAMING_OK` to treat a packet as valid. At the moment I am not sure whether the problem is the parser, the log or both.

Arne-W, 2017-11-27:
I did several tests and checked mavlink just to be sure there was no bigger change I missed. But even with the current master of the mavlink code the APM-Planner won't parse the data out of this file. Always same result - exactly one message. At the moment I tend to say the log is broken.... But we should do more investigation.

@AndKe Did you test the last changes? Planner should not crash when opening this files and it should be possible to view the logs with more than one flight...

AndKe, 2017-11-27:
@Arne-W Oh yes, the display of multi-flight logs is great.

As for the above file, it's generated by mavproxy, and - using mavlogdump.py works very well, I would think mavproxy (I build myself from master) also uses very fresh mavlink code ?

Arne-W, 2018-06-28:
@AndKe Seems like this is fixed due to the mavlink 2 migration we are doing here #1130. "Battery_status.." messages have an error cause they lack a voltages field - don't know why.

Arne-W, 2018-07-23:
Seems to be fixed by
284b0fdb737da405a21b49c857643b6401ddedeb and 736532f14d6dd429f860627783936cceaae971ec

closed 2018-07-23 by Arne-W