ArduPilot/apm_planner #1077: Time for a README.md update?

drobban, 2017-11-11:
First time submitting an issue and I'm not sure that my report belong here.

During the time I have been waiting for my quadcopter building material to arrive, I have spent some time reading and testing apm planner2 with simulator. 

In this process I have relied on much of the information provided in the README and discovered at least one broken link and a suggestion for improvement.

Link that is broken is: 
http://planner2.ardupilot.com I suspect that the correct is: http://ardupilot.org/planner2/

billbonney, 2017-11-11:
fixed, thx 👍

closed 2017-11-11 by billbonney

drobban, 2017-11-11:
hehe :) I hoped that I could get to make a pull request for that one...
Well, perhaps I can find something else in the future to improve.

billbonney, 2017-11-11:
@drobban Sure, please make a PR and we can review and commit the changes. Contributions are more than welcome. 😄