ArduPilot/apm_planner #1080: make install fails

gusisin, 2017-11-17:
I am running debian .....

_[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.4.39-h3 (root@wwd-ubuntu) (gcc version 4.6.3 20120201 (prerelease) (crosstool-NG linaro-1.13.1-2012.02-20120222 - Linaro GCC 2012.02) ) #65 SMP PREEMPT Thu Sep 1 13:44:36 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: sun8i
[    0.000000] cma: CMA: reserved 256 MiB at 50000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 262144_

I follow the install steps and I get the following message from the compiler:

_**root@pcDuino:/home/ls/mavlink/apm_planner# make install
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIE -DQ_LINUX_32 -DFLITE_AUDIO_ENABLED -D_TTY_NOWARN_ -D__STDC_LIMIT_MACROS -DGIT_COMMIT=2.0.25-23-g9910ae2 -DGIT_HASH=9910ae2e7e77ef9d725f1d3cbb6276c6d005fea0 -DOPENSSL -DQT_NO_DEBUG -DMAVLINK_NO_DATA -DQGC_USE_ARDUPILOTMEGA_MESSAGES -DNOMINMAX -DEXTERNAL_USE -DDATADIR=\"/share\" -DPKGDATADIR=\"\" -DQT_NO_DEBUG -DQT_QUICKWIDGETS_LIB -DQT_QUICK_LIB -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_OPENGL_LIB -DQT_QML_LIB -DQT_WIDGETS_LIB -DQT_TESTLIB_LIB -DQT_SERIALPORT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_NETWORK_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR=\"/home/ls/mavlink/apm_planner\" -I/extended/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -Ilibs/mavlink/include/mavlink/v1.0 -Ilibs/mavlink/include/mavlink/v1.0/ardupilotmega -Ilibs/alglib -Ilibs/opmapcontrol/src/core -Ilibs/opmapcontrol/src/internals -Ilibs/opmapcontrol/src/internals/projections -Ilibs/opmapcontrol/src/mapwidget -Ilibs/utils -Ilibs -Ilibs/opmapcontrol -Ilibs/thirdParty/quazip -I/usr/src/3rdparty/zlib -I. -Isrc -Isrc/ui -Isrc/ui/linechart -Isrc/ui/uas -Isrc/ui/map -Isrc/uas -Isrc/comm -Iinclude/ui -Isrc/input -Isrc/lib/qmapcontrol -Isrc/ui/mavlink -Isrc/ui/param -Isrc/ui/watchdog -Isrc/ui/map3D -Isrc/ui/mission -Isrc/ui/designer -Isrc/ui/configuration -Isrc/output -I/extended/usr/include/arm-linux-gnueabihf/qt5 -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtQuickWidgets -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtQuick -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtPrintSupport -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtSvg -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtQml -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtTest -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtSql -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtXml -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtNetwork -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtGui -I/extended/usr/include/arm-linux-gnueabihf/qt5/QtCore -Ibuild-release/moc -Ibuild-release/ui -o build-release/obj/main.o src/main.cc
src/main.cc: In function ‘void loggingMessageHandler(QtMsgType, const QMessageLogContext&, const QString&)’:
src/main.cc:70:64: error: ‘qFormatLogMessage’ was not declared in this scope
     QString outMessage(qFormatLogMessage(type, context, message));  // just format only once
                                                                ^
Makefile:73859: recipe for target 'build-release/obj/main.o' failed
make: *** [build-release/obj/main.o] Error 1**_

This is the line it complains about.

62 void loggingMessageHandler(QtMsgType type, const QMessageLogContext &context, const QString &message)
 63 {   
 64     // The message handler has to be thread safe
 65     static QMutex mutex;
 66     QMutexLocker localLoc(&mutex);
 67     
 68     static std::ofstream logFile(sLogPath.toStdString().c_str(), std::ofstream::out | std::ofstream::app);
 69 
 70     QString outMessage(qFormatLogMessage(type, context, message));  // just format only once
 71     if(logFile)
 72     {   
 73         logFile << qPrintable(outMessage) << std::endl; // log to file
 74     }
 75     
 76     LogWindowSingleton::instance().write(outMessage);   // log to debug window
 77     
 78     fprintf(stderr, "%s\n", qPrintable(outMessage));    // log to console
 79 }

I looked at the makefile and could see no obvious ommision. Can you suggest?

drobban, 2017-11-18:
just to make sure.
What version of Qt is installed on your machine?

Arne-W, 2018-03-02:
@gusisin 
This method was introduced in Qt 5.4
`QString qFormatLogMessage(QtMsgType type, const QMessageLogContext &context, const QString &str)`

closed 2018-03-02 by Arne-W