ArduPilot/apm_planner #1092: Graph: fail to handle a small  .BIN

AndKe, 2018-02-14:
please try to graph see attached file.
AP2 will be stuck with an empty infobox.

[00000018.zip](https://github.com/ArduPilot/apm_planner/files/1724242/00000018.zip)

a quick 
mavlogdump.py 00000018.BIN | wc -l

confirms that there are 7682 lines...

Arne-W, 2018-02-14:
@AndKe For me everything works as expected. The log was parsed without any errors resulting in 7552 loglines + 128 FMT lines = 7681 which matches your expectation - the "off by one" can be ignored I think. I have used the current master.
So from my point of view there is no issue 😉

AndKe, 2018-02-14:
true, unable to reproduce anymore. Sorry.
What happened, is that I had another log open, opened it , it failed with a small empty box, I tried to open it again.. reported.

closed 2018-02-14 by AndKe

Arne-W, 2018-02-15:
No problem - always glad to help you 😄