ArduPilot/apm_planner #1095: PX4FirmwareUploader.cc - Compile error

drobban, 2018-02-22:
**Background**
I'm working on an port of apm_planner to OpenBSD and any input you have about this is much appreciated as my ability to test this code is limited.
But I'm able to "test" by compiling in both Linux and OpenBSD.

**Environment**
Compiler: clang
OpenSSL: LibreSSL 2.6.3
OS: OpenBSD 6.2
apm_planner: latest commit https://github.com/ArduPilot/apm_planner/commit/8b66bf1d4dcc597937ccaae20421603f338a24fc

*Where*
https://github.com/ArduPilot/apm_planner/blob/master/src/ui/configuration/PX4FirmwareUploader.cc#L654

*Why*
missing declaration of EVP_PKEY_get0_RSA.

*Non confirmed solution*
There is however a EVP_PKEY_get1_RSA declared in <openssl/evp.h>
after changing that I was able to build a working version of apm_planner in OpenBSD 6.2

**Question**
What potential consequences does my fix have?
Do you have any other proposed solutions that can be applied in both OpenBSD, Linux & others?

Arne-W, 2018-02-22:
Looks like it is used to verify the downloaded FW image. Is it still possible to verify a valid FW with the new function?

This is what I found somewhere in the net but I don't know what they mean incrementing the ref count?
-----------------
Notice that the ref count is updated. There are then two getters:

RSA *EVP_PKEY_get0_RSA(EVP_PKEY *pkey)
RSA *EVP_PKEY_get1_RSA(EVP_PKEY *pkey)

Where EVP_PKEY_get1_RSA will call EVP_PKEY_get0_RSA and then increment the ref count. This is the only reason I can think of that these function have 1 and 0. 1 for functions that update the ref count and 0 for those that dont. "In accordance with the OpenSSL naming convention the key obtained from or assigned to the pkey using the 1 functions must be freed as well as pkey.
-----------------

drobban, 2018-02-22:
Thank you @Arne-W for the reply.

So if I understand this correctly:
In order to fully replace EVP_PKEY_get0_RSA with EVP_PKEY_get1_RSA
a equivalent solution would be to move the getter outside on previous row, get the key and after the verification on row 654 free it with a call to ` void RSA_free(RSA *rsa);`

something like this

`RSA * rsa_key = EVP_PKEY_get1_RSA(pkey);`

`int verify = RSA_verify(NID_sha1,(unsigned char*)serial.data(),serial.size(),(unsigned char*)signature.data(),signature.size(), rsa_key);`

`RSA_free(rsa_key);`

**LibreSSL**
What I can see and understand, the get0 function isn't available in the libssl library.
The suggestion I made, is that something you would accept a pull request on? What I understand the functionality shouldn't affected (but this is nothing I have knowledge on how to test).

Another way could be to add #ifdef guard and only use "this" on OpenBSD platform and leave the above code as is.

It would be nice if apm_planner upstream added support for OpenBSD. Im in early stage, but the program seems to work just fine in OpenBSD.

Arne-W, 2018-02-22:
Hmm - if supporting OpenBSD does not add to much `#ifdef` and `#ifndef` I do not see any reason why we should not make it buildable on OpenBSD. You know supporting means anyone in the team will care about bugs which happen only on OpenBSD and provides new builds for the platform - and I think we do not have someone who will / can do it. @billbonney Is there any reason why we should not make APM-Planner buildable on OpenBSD??

Regarding your workaround from above you should check the last commit on this file [https://github.com/ArduPilot/apm_planner/commit/780acb522032458ee5b5a83536e76746e254609b](url) and perhaps contact the author @larsks about that change - looks like it could fail on other plattforms?!

larsks, 2018-02-22:
@Arne-W I probably don't have anything constructive to add here: I'm not really familiar with openssl; I was simply implementing a fix that had been recommended elsewhere.

drobban, 2018-02-22:
I tried to study the manuals that I can find.
To my understanding, both older and newer versions of ssl have a get1 function declared, In some cases there is a get0 missing.

Both get1 and get0 "returns" the same kind and content. The data of interest is only used in the verify function.

There should not be a problem replacing get0 with get1 instead and skip the problem with missing declarations. But, I don't have PX4 hardware so I'm unable to verify that it in fact actually work.

Would of been nice with a unit test :-)

Regarding OpenBSD support;
All this is an effort from my side to bring the drone tools I use in Linux to the OpenBSD platform, so I'm more then willing to help out maintaining the OpenBSD support.

Arne-W, 2018-02-23:
Ahh ok nice to hear that. So make the changes and create a PR. As I have a PixHawk which needs to be updated I can test and review the change then. If it works I don't know any reason why should not pull it in.

@larsks Nevertheless thanks for the info!

billbonney, 2018-02-24:
You can drop the Certificate Of Authenticity check. It was a feature to verify boards from 3DR etc... It's not really used, if at all anymore.

Welcome a port to OpenBSD, there shouldn't be too many incompatibilities as it handled by Qt mainly (expect this one)

Arne-W, 2018-02-24:
If we can completely remove the `PX4FirmwareUploader::checkCOA(...)` function we should do it! @drobban that would be the easiest way to solve your issue I think :smiley:

drobban, 2018-02-24:
Great to hear.
I'll look into the removal of `checkCOA(...)`, I have another issue that I need to resolve as well.
Looks like the mavlink over udp communication is broken in my OpenBSD build.
If it is okey with you guys I would like to make one pull request that fixes all functionality issues for OpenBSD.

billbonney, 2018-02-27:
@drobban sure, go ahead

Arne-W, 2021-04-09:
checkCOA(...) has been removed long time ago.... Therefore I will close this issue now.

closed 2021-04-09 by Arne-W