ArduPilot/apm_planner #1097: Please add flight timer.

AndKe, 2018-02-23:
I am attaching a piece of MAVProxy code to show how it can be done across platforms...

```
           flying = False
            if self.mpstate.vehicle_type == 'copter':
                flying = self.master.motors_armed()
            else:
                flying = msg.groundspeed > 3
            if flying and not self.in_air:
                self.in_air = True
                self.start_time = time.mktime(t)
            elif flying and self.in_air:
                self.total_time = time.mktime(t) - self.start_time
                self.console.set_status('FlightTime', 'FlightTime %u:%02u' % (int(self.total_time)/60, int(self.total_time)%60))
            elif not flying and self.in_air:
                self.in_air = False
                self.total_time = time.mktime(t) - self.start_time
                self.console.set_status('FlightTime', 'FlightTime %u:%02u' % (int(self.total_time)/60, int(self.total_time)%60))
```

Arne-W, 2018-02-23:
Where do you want it? In a special view or somehow in the menu bar - perhaps in the gap between the mode and the serial parameters? Perhaps like this:
[image]

billbonney, 2018-02-23:
You should create widget that can be added to the desktop, or add it to the unmanned system widget.

Or you can publish it as a parameter to show in the InfoView

Or add it to the PFD

Some suggestions

Arne-W, 2018-02-24:
@billbonney seems you are the best one to implement - having those great ideas :wink: :smiley:

AndKe, 2018-02-24:
I think this kind of info kind of belong in Info "View/Quick"  tab. 
However, Other essential info, like voltage and throttle level could be placed a more prominent place, like the mockup from @Arne-W or in PFD

billbonney, 2018-02-24:
You could add it to the MAP. In anywise, i can help somebody with how to implement a widget if required, but I don't have time to do it myself at the moment