ArduPilot/apm_planner #1099: Graph: Error Message status is confusing

AndKe, 2018-03-02:
I will sum up what AC dataflashlog MSG says , then what AP2 display in graph:

"GPS Glitch"  displayed as  "GPS: Unknown ErrorCode(2)"   (confusing at best)
"EKF primary changed:1"  becomes "EKF primary: Failed to init"  (sounds far worse than it is ?)
"GPS Glitch Cleared" becomes "GPS:Everything OK!"    (correlation between glitch (errorcode) and glitch cleared is completely gone.)

may I suggest to stop "translating" the strings, and just present them as-is ?  - for an end user, it's much more likely to find anything about a dataflashlog message, than a "rephrased" AP2 message.

Arne-W, 2018-03-02:
Hi @AndKe 
the message formatter does no translating. The error messages are transmitted as IDs which are converted into text to give the user an idea of what is appening. Therefore the I have to update the formatter as soon as there is a new message - in case of the GPS unknown ErrorCode it is a new (7 month old) code and the last update of the formatter was done in 01.2017 so it is unknown to the formatter.
In case of the EKF primary the Errorcode 1 ***is*** defined as Failed to initialize.
The Everything OK message is generated due to the fact that there is no code defining that the (new) glitch is gone. There is only a general code describing that all errors are resolved which results in the "Everything OK!" message.

Take a look [ArduCopter/defines.h](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/defines.h#L417) there you can find the Ids the messages are generated from. And here [src/uas/ArduPilotMegaMAV.cc](https://github.com/ArduPilot/apm_planner/blob/master/src/uas/ArduPilotMegaMAV.cc#L833) you can see how they are generated.

I never get informed if someone changes the IDs so I have to poll those files - looks like it is time to do a check what has changed....

AndKe, 2018-03-02:
You are right (of course)
First there was this line
ERR subsys 11  ecode 2 
then the MSG "GPS Glitch" right after :)  - which made me believe what i saw was another way to present the MSG

Arne-W, 2018-03-04:
@AndKe I have updated the formatters - `ERR subsys 11 ecode 2` should now resolve to `GPS: Glitch detected`

closed 2018-03-04 by Arne-W

AndKe, 2018-03-04:
nice, and SmartRTL and other goodies spawned as well :)
Thanks.

Arne-W, 2018-03-05:
Always welcome - you can always inform me if you see messages which show only and ID. For Copters there should be always a "translation" if not the ID be new.

commit 854d6e7e (2018-03-04): Graph View: Adapted the message Formatters to new messages. Fixes #1099

Added Error message GPS:Glitch detected
Added Mode message Smart RTL and Flowhold, Mode reason Flight termination and Toy mode
Added Event message Primary GPS changed, Winch relaxed, Winch length control and Winch rate control
commit f07a5a03 (2018-03-04): Graph View: Adapted the message Formatters to new messages. Fixes #1099

Added Error message GPS:Glitch detected
Added Mode message Smart RTL and Flowhold, Mode reason Flight termination and Toy mode
Added Event message Primary GPS changed, Winch relaxed, Winch length control and Winch rate control