ArduPilot/apm_planner #1101: Graph: reset cursors on x-axis change

AndKe, 2018-03-02:
To reproduce, 
1 load a graph
2 check "show table cursor" (or range cursor)
3 check "use time on x-axis"
=Observe that the cursor(s) are gone, while checkbox(es) are checked.

One simple workaround would be to uncheck table cursor and range cursor checkboxes on x-axis type change.
a more complex solution would be to make the cursors survive such change.

Arne-W, 2018-03-02:
Ah - The cursors are removed but looks like I forgot to disable the checkboxes. Should be easy to solve - perhaps we can make them survive the change - I will have a look

Arne-W, 2018-03-17:
@AndKe Seems to be not very easy to make them survive the axis change. The data model lacks a method to convert a index position to a valid time stamp. As the whole view is reset on axis change I think we should go the easy way - don't think many people will miss the "surviving cursors".
What do you think?

AndKe, 2018-03-17:
Both linenumber->time and time->linenumber convertions could be based on the fact that "simple cursor" position translate well to a log-line.  Wherever I drop the simple cursor, proper line in the grid is selected.

If you could look up the proper line using that method for each cursor, it would work.
What I do not know, if that's a lot of work, or less.

To me the cursors surviving is not essential at all, I would rather propose you spend your valuable time on "cooler" features, if you consider them worthy/interesting:

- store and use last used x-axis mode (I , most often, use time on X axis) 
- in the tooltip over graph, display GPS date/time (based on nearest GPS message for dataflashlogs) , or, for .tlog ,try to parse the filename for datetime, and offset by time_boot_ms (plus previous max times? in tlog)  .. that would be hard, maybe just GPS time for dataflashlogs..  :)
-third idea: let apmplanner2 try to graph any log specified as argument when starting (enable us to assin .bin to be opened using AP2) 
 
...and so I derailed this thread from it's original topic :)

Arne-W, 2018-03-19:
@AndKe Thats what I thought. No need to implement features perhaps nobody needs. There are a lot things to do. So for now I will choose the simple solution - disable checkbox when cursor is removed.
BTW the x-axis mode storing will be in my next PR :wink:

commit 28b2e8b2 (2018-04-01): Graph View: Cursor check boxes will be unchecked when changing x-axis. Fixes #1101
closed 2018-04-01 by Arne-W