ArduPilot/apm_planner #1106: Unable to parse log

liang-tang, 2018-03-06:
APM planner version： 2.0.24
ardupilot version：copter 3.8.4

[image]

Arne-W, 2018-03-06:
@liang-tang I strongly recommend to update your APM-Planner version. I would recommend to use 2.0.26-rc1 which can be found [here](http://firmware.ardupilot.org/Tools/APMPlanner/beta/), or look in the [discussion forums](https://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0) there are several topics containing a 2.0.25 version for various plattforms.
Or build from scratch by yourself.
Between 2.0.24 and 2.0.25 the log parsing has been completely rewritten and 2.0.26 adds a lot of bugfixes to several parts of the APM-Planner

liang-tang, 2018-03-06:
@Arne-W Thanks very much，I have update apm-planner to 2.0.26

closed 2018-03-06 by liang-tang

Arne-W, 2018-03-06:
:+1: that version should work like a charm... :smiley: