ArduPilot/apm_planner #1107: Graph/Export: .kml export success, then segfault

AndKe, 2018-03-06:
```
Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007ffff5d33728 in QDialog::accept() () from /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Widgets.so.5

```
- the exported file is fine. AP2 crashes once I click "OK" inside the "success" messagebox.
- I am not sure whatever the file I attached is required to reproduce. 
- 100% reproducible every time here.
- could somehow be related to Qt5.9.4 , which is otherwise working perfectly (?)

[2018-03-02 14-29-16.zip](https://github.com/mavlink/qgroundcontrol/files/1786195/2018-03-02.14-29-16.zip)

Arne-W, 2018-03-06:
@AndKe Tested with Qt 5.6.2 without any issue. But I have an Idea what is happening - I suppose the crash is not log related. Seems like the order of object destruction is the culprit which is handled in a too optimistic way.
If you comment the creation of the messagebox [https://github.com/ArduPilot/apm_planner/blob/master/src/ui/Loghandling/LogExporter.cpp#L194](https://github.com/ArduPilot/apm_planner/blob/master/src/ui/Loghandling/LogExporter.cpp#L194) the crash won't happen. Would be cool if you could test it :smiley:

Arne-W, 2018-03-06:
@AndKe I made a fix - would be nice if you could test it. You can find the branch in my fork [https://github.com/Arne-W/apm_planner/tree/BugFix_kml_export](https://github.com/Arne-W/apm_planner/tree/BugFix_kml_export) Hopefully this fixes the issue.

AndKe, 2018-03-06:
Tested the fix:
'Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x000055555a307ed0 in ?? ()'

and if I also comment out the messagebox (on your fix branch) I get:
Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x0000000000000040 in ?? ()

Arne-W, 2018-03-06:
Damn - seems to be not that easy. I will have to install Qt5.9.4 and try to reproduce it. Thanks for testing so far. Could you please send the last lines (10) of the log file? Perhaps it can give us a hint.

AndKe, 2018-03-06:
the log does not look promising, both master and your fix build end with the "Plot Log" line.
```

[20180306 22:47:29.733 INFO ] - setting objectName:  "VIEW_SIMULATOR"
[20180306 22:47:29.897 INFO ] - UASActionsWidget creating  UASActionsWidget(0x7fffe0005460)
[20180306 22:47:29.955 INFO ] - Using Alsa Audio driver
[20180306 22:47:29.955 WARN ] - Dir directory tmp_audio exists
[20180306 22:47:29.965 INFO ] - NETWORK_PROXY: Disabling System Network Proxies
[20180306 22:47:30.063 WARN ] - QSettings::endGroup: No matching beginGroup()
[20180306 22:47:30.093 INFO ] - "ttyS4" 115200
[20180306 22:47:30.093 INFO ] - UDP Created  "UDP Link (port:14550)"
[20180306 22:47:30.093 INFO ] - UDPLink::UDP connect  QHostAddress(QHostAddress::Any) : 14550
[20180306 22:47:46.160 INFO ] - Plot Log loading took 1.788 seconds - 17713554 of 17713554 bytes used
```

Arne-W, 2018-03-07:
Could it be possible that you somehow defined `NO_DEBUG` because the highest loglevel in your log is `INFO`. There should be some logmessages about constructing and destructing the exporter objects and it would be nice to see which object ist destructed right before the crash.

AndKe, 2018-03-07:
no, I just downloaded and compiled it, nor do I have such system variable.  Maybe there were no errors ? :)

Arne-W, 2018-03-07:
Just installed Qt5.9.4 build from scratch and can not reproduce it.
What makes me still wondering is the Log level. In my log there are many `DEBUG` messages which are all missing in your log. The only way to switch them off is to define the `NO_DEBUG` symbol. This is what happens between your  last 2 log lines:
```
[20180307 18:46:47.576 INFO ] - UDPLink::UDP connect  QHostAddress(QHostAddress::Any) : 14550
[20180307 18:46:47.589 DEBUG] - SCW: Link is disconnected
[20180307 18:46:47.597 DEBUG] - Baud rate: 115200 Expected: 4
[20180307 18:46:48.558 DEBUG] - Writing ( "http://autotest.ardupilot.org/Parameters/ArduPlane/apm.pdef.xml" ) to ( "/home/blaster/apmplanner2/ardupilotmega/arduplane.pdef.xml" )
[20180307 18:46:49.621 DEBUG] - Writing ( "http://autotest.ardupilot.org/Parameters/ArduCopter/apm.pdef.xml" ) to ( "/home/blaster/apmplanner2/ardupilotmega/arducopter.pdef.xml" )
[20180307 18:46:50.462 DEBUG] - Writing ( "http://autotest.ardupilot.org/Parameters/APMrover2/apm.pdef.xml" ) to ( "/home/blaster/apmplanner2/ardupilotmega/ardurover.pdef.xml" )
[20180307 18:46:51.111 DEBUG] - APMToolBar: GRAPHS SELECTED
[20180307 18:46:51.112 DEBUG] - APMToolBar: clear selected buttons
[20180307 18:46:51.112 DEBUG] - APMToolBar: FLIGHT DATA unselected
[20180307 18:46:51.113 DEBUG] - APMToolBar: GRAPHS unselected
[20180307 18:46:51.306 DEBUG] - APMToolBar: GRAPHS SELECTED
[20180307 18:46:52.484 DEBUG] - "retrieve versionobject from server: http://firmware.ardupilot.org/Tools/APMPlanner/apm_planner_version.json"
[20180307 18:46:52.722 DEBUG] - Downloading: 1147 / 2838
[20180307 18:46:52.723 DEBUG] - Downloading: 2838 / 2838
[20180307 18:46:52.723 DEBUG] - AutoUpdateCheck::httpFinished()
[20180307 18:47:05.391 DEBUG] - LogAnalysis::LogAnalysis - CTOR
[20180307 18:47:05.396 DEBUG] - PresetManager::PresetManager - CTOR
[20180307 18:47:05.408 DEBUG] - LogAnalysis::loadLog - Filename: "/home/blaster/blaster-nas/develop/apmPlanner/testLogs/2018-03-02 14-29-16.log"
[20180307 18:47:05.408 DEBUG] - LogdataStorage::LogdataStorage()
[20180307 18:47:05.408 DEBUG] - Created AP2DataPlotThread: AP2DataPlotThread(0x62cddf0)
[20180307 18:47:05.410 DEBUG] - AP2DataPlotThread::run(): Log loading start - 17713554 bytes
[20180307 18:47:05.410 DEBUG] - LogParserBase::LogParserBase - CTOR
[20180307 18:47:05.410 DEBUG] - AsciiLogParser::AsciiLogParser - CTOR
[20180307 18:47:05.410 DEBUG] - AsciiLogParser::parse: "/home/blaster/blaster-nas/develop/apmPlanner/testLogs/2018-03-02 14-29-16.log"
[20180307 18:47:08.614 DEBUG] - AsciiLogParser::AsciiLogParser - DTOR
[20180307 18:47:08.614 DEBUG] - LogParserBase::LogParserBase - DTOR
[20180307 18:47:08.614 INFO ] - Plot Log loading took 3.204 seconds - 17713554 of 17713554 bytes used
```
and this is what happens on a KML export:
```
[20180307 18:47:14.414 DEBUG] - Suggested Export Filename:  "/home/blaster/blaster-nas/develop/apmPlanner/testLogs/2018-03-02 14-29-16.kml"
[20180307 18:47:28.823 DEBUG] - iconInterval:  2
[20180307 18:47:28.824 DEBUG] - LogExporterBase::LogExporterBase()
[20180307 18:47:28.825 DEBUG] - KmlLogExporter::KmlLogExporter()
[20180307 18:47:28.825 DEBUG] - LogExporterBase::exportToFile() Filename: "/home/blaster/Schreibtisch/2018-03-02 14-29-16_2.kml"
[20180307 18:47:28.845 DEBUG] - MAV_TYPE:  2 , flight mode:  0 :  "Stabilize" , color:  "FF00FF00"
[20180307 18:47:28.848 DEBUG] - MAV_TYPE:  2 , flight mode:  0 :  "Stabilize" , color:  "FF00FF00"
[20180307 18:47:28.899 DEBUG] - MAV_TYPE:  2 , flight mode:  3 :  "Auto" , color:  "FFFF00FF"
[20180307 18:47:31.230 DEBUG] - MAV_TYPE:  2 , flight mode:  16 :  "Pos Hold" , color:  "FF00F000"
[20180307 18:47:31.423 DEBUG] - write kml to  "/home/blaster/Schreibtisch/2018-03-02 14-29-16_2.kml"
[20180307 18:47:31.713 DEBUG] - Done
[20180307 18:47:31.713 DEBUG] - "Successfull exported to /home/blaster/Schreibtisch/2018-03-02 14-29-16_2.kmz."
[20180307 18:47:38.693 DEBUG] - KmlLogExporter::~KmlLogExporter()
[20180307 18:47:38.695 DEBUG] - LogExporterBase::~LogExporterBase()
[20180307 18:47:38.696 DEBUG] - Log export took  9875 ms
```

AndKe, 2018-03-07:
yes, this is intriguing.  :)
the build was made by downloading your branch, qmake, them make..
I checked 
set | grep DEBUG
also .profile & .bashrc 
can't find anything about NO_DEBUG

AndKe, 2018-03-07:
also: 
```
andre@loke:~/Downloads/apm_planner-BugFix_kml_export$ grep -inr NO_DEBUG
Makefile:15:DEFINES       = -DQ_LINUX_32 -DFLITE_AUDIO_ENABLED -DQ_UBUNTU -D_TTY_NOWARN_ -D__STDC_LIMIT_MACROS -DGIT_COMMIT= -DGIT_HASH= -DOPENSSL -DQT_NO_DEBUG -DMAVLINK_NO_DATA -DQGC_USE_ARDUPILOTMEGA_MESSAGES -DNOMINMAX -DEXTERNAL_USE -DDATADIR=\"/share\" -DPKGDATADIR=\"\" -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_QUICKWIDGETS_LIB -DQT_WIDGETS_LIB -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_SQL_LIB -DQT_SERIALPORT_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"/home/andre/Downloads/apm_planner-BugFix_kml_export"'
apm_planner.pro:276:    DEFINES += QT_NO_DEBUG
src/logging.h:26://#define NO_DEBUG
src/logging.h:39:#ifdef NO_DEBUG
```

Arne-W, 2018-03-08:
The -D means define, so -DQT_NO_DEBUG means define symbol QT_NO_DEBUG while building - but thats default if doing a release build. Perhaps I shoult try a console build. I always build from my IDE - perhaps it is doing some magic.... Just to be sure you build from scratch?

AndKe, 2018-03-08:
>>Just to be sure you build from scratch?
not sure what other alternatives there are:
I downloaded you branch as a zip, then:
qmake qgroundcontrol.pro  && make -j8

Arne-W, 2018-03-08:
I am sorry - I can't reproduce it. But I have to do it in a slightly different way to use Qt5.9.4. I have to call:
`/opt/Qt/5.9.4/gcc_64/bin/qmake -o Makefile apm_planner.pro && make -j4 ` otherwise it uses my system Qt. Perhaps you should try this way? You could also try `/opt/Qt/5.9.4/gcc_64/bin/qmake -o Makefile apm_planner.pro -spec linux-g++-64 && make -j4` I am not sure if it really has an effect, or if it is only from old times.
If all this does not work you could upload the generated makefile I could do some comparing - don't know it it helps.

AndKe, 2018-03-08:
you specify qmake path because it's not in your path, mine is, so that explains most of the difference:
```
$ which qmake
/home/andre/Qt/5.9.4/gcc_64/bin/qmake
``` 

rebuilding it with all the extra options does not seem to change anything, not a single debug message in the console.
Attaching Makefile
 
[Makefile.zip](https://github.com/ArduPilot/apm_planner/files/1794452/Makefile.zip)

drobban, 2018-03-08:
Not sure if my observations will help out.
But here they are;
As the problem and behavior you are experiencing is dist/OS dependent I'm going to list what  I'm running:
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

To get the debug messages to show
(make clean to be sure)
qmake -config debug apm_planner.pro
make

(the newly built files are found in apm_planner/debug/apmplanner2

This to get rid of -DQT_NO_DEBUG - Read http://doc.qt.io/qt-5/qtglobal.html#qDebug

But for me this wasn't enough, I also had to make a qtlogging.ini file inside .config/QtProject/
My config looks like this; 
[Rules]
*.debug=true
qt.*.debug=false

Now you should get some pretty debug messages all over the terminal...

This is very confusing as it seems to be system dependent. In OpenBSD I have to #define NO_DEBUG to get rid of the messages.

AndKe, 2018-03-08:
@drobban Thank you.
I am also on Ubuntu 17.10   - It's getting late, so I did not change the .ini
for now, I just built it with "-config debug" while replying, and that alone did not fix it, yet I am sure that the .ini will do the trick.  - quite messy stuff, I'd expect not to have to change the QT settings to compile something as the Makefile / .pro file says.

drobban, 2018-03-09:
I'm not certain, but I believe the .ini file isn't used in compile-time, instead I think it is used by the Qt run-time to determine if the debug-stream should be silenced or not...

It is messy, but also kind of neat that someone is able to change the log-level as things tends to drown in the overflow of information. But I said earlier, not sure if it really is a run-time variable, as I found out about this yesterday when I tried to solve this mystery

AndKe, 2018-03-09:
based on the lines in logging.h
```
//#define NO_DEBUG
.
.
.
#ifdef NO_DEBUG
    #define QLOG_DEBUG() if(1){} else qDebug()
#else
    #define QLOG_DEBUG() qDebug()
#endif
```

I really consider this some sort of bug, the fact that logging is not reproducible on all systems, after being defined in the code.

drobban, 2018-03-09:
Yes and no?
The fact is that the code is compiled into the binary. But the Qt manual states that, if QT_NO_DEBUG is defined it will "deactivate" the qDebug output. So the "problem" is not in the pre-process step. Perhaps misleading, that would be the result of some things that happen in compilation and some in run-time.

And why Qt also uses the `qtlogging.ini` is a mystery, it does not in OpenBSD and it does not care about QT_NO_DEBUG in OpenBSD, but that isnt related to the apm_planner code. Perhaps its a bug, but not in the apm_planner code as far I understand and know.

drobban, 2018-03-09:
However, something that would make things more clear is if `#ifdef NO_DEBUG` was replaced with `#ifdef QT_NO_DEBUG`. As things are now, is that we have 2 DEBUG defines doing almost the same things.

Arne-W, 2018-03-09:
The idea of this macros is to remove the code at compile time. If `NO_DEBUG` is defined the `QLOG_DEBUG()` macro resolves to the somehow useless `if(1){} else qDebug()` code which is optimized away by the compiler. 
I used this technique when I rebuild the loggingsystem - perhaps it was not a good idea cause the Qt logging does the same thing somewhere else. I will check that.

Arne-W, 2018-03-09:
@drobban no we did not want to remove debug output in release - it very heplful if people are having problems.

drobban, 2018-03-09:
well @Arne-W correct me if I'm wrong. But the debug output is silenced by QT_NO_DEBUG as is right now. At least for me and what it seems for @AndKe and what the manual says, that is the expected result.

The "practical" difference is that NO_DEBUG removes qDebug from compilation, something I think it should do, :+1: as this removes debug messages for me in OpenBSD.

What I'm suggesting is but might have a problem:
Use QT_NO_DEBUG instead of NO_DEBUG and make sure that the QT_NO_DEBUG -flag isnt used in the release build. And here is the problem, when I qmake apm_planner.pro it automaticly adds QT_NO_DEBUG even if its not added in the .pro file.

Arne-W, 2018-03-09:
I searched a bit in the and it looks like we doubled the mechanics. The `QT_NO_DEBUG` flag seems to do the same as the `NO_DEBUG` but in an other way.
What still confuses me is the fact that the Qt mechanism (unsing `QT_NO_DEBUG`) does not work on Ubuntu 16.04 even when I use Qt5.9.4
Perhaps we the new code should switch to categorized logging which enables truning on an off logging messages at runtime. The macros could be left for compatibiltity reasons otherwise the refactoring would be too much work I think.
We could remove the macro switching for all loglevels except for trace - as this is mapped to debug we can not differentiate between debug an trace.

@billbonney Perhaps you have a better idea?

Arne-W, 2018-03-09:
But we should focus - on the crash induced by the kml export - the Logging issue should be discussed in #1108
@AndKe please use this `qmake apm_planner.pro CONFIG+=debug` if this does not help it must be somehow related to your system. Nevertheless I will try to setup a VM and try to reproduce.

Arne-W, 2018-03-10:
After several attempts :confused: I have a working VM with Ubuntu 17.10 :smiley: ....
But I can not reproduce the crash - tried several times and every export worked!
The Logging issue was reproducible.

Arne-W, 2018-03-12:
@AndKe As we now know how to enable the logging in 17.10, it would be nice if you could post a log with this set of logging rules:
```
[Rules]
.debug=true
qt..debug=false
```
and one with this set:
```
[Rules]
.debug=true
```
Both should end with the crash - perhaps we can find what is different to my setup and might lead to this ugly crashing.

Arne-W, 2018-03-13:
@AndKe Could is be possible that you posted the log in the wrong issue? :wink: 
Nevertheless I will answer here. Somehow your application crashes in the accept method of the dialog. It is really strange - does it also happen if you do an ASCII export? ASCII and KML export use mainly the same functionality - if it is related to the exporter dialog handling both exports should behave the same. Moreover you could try this branch [https://github.com/Arne-W/apm_planner/tree/logging_test](https://github.com/Arne-W/apm_planner/tree/logging_test). It should fix the logging - you should have debug output regardless what is set in you `qtlogging.ini` and it changed the dialog handling once again - the dialog is opened from another parent which is not destroyed at the end onf the export.

Arne-W, 2018-03-14:
Thats annoying.... Does it also happen if you comment this line: [https://github.com/Arne-W/apm_planner/blob/logging_test/src/ui/Loghandling/LogAnalysis.cpp#L1212](https://github.com/Arne-W/apm_planner/blob/logging_test/src/ui/Loghandling/LogAnalysis.cpp#L1212)?
...running out of ideas....somehow - but not completely :smiley:

AndKe, 2018-03-14:
Commented out , no message, but crash right after successful export:
```
[20180314 17:23:13.663 DEBUG] - Log export took  1521 ms
Segmentation fault (core dumped)
```

Arne-W, 2018-03-15:
To summarize: It is not related to the type of export (ascii / kml) and it is not related to the end of export dialog.
The last thing that is left in this kontext is the filedialog which is used with the signal / slot mechanism which does not seem to be good style (like it is done here). Perhaps I can do a fast hack on next weekend to make a proof.

Arne-W, 2018-03-16:
@AndKe Changed the file dialog handling. If this does not help I am out of ideas. All objects are local now and AFAIK used as they should - no magic, clean and good style. Its the same branch as [above](https://github.com/ArduPilot/apm_planner/issues/1107#issuecomment-372776178)
I'll keep my fingers crossed!

AndKe, 2018-03-16:
you did it again :1st_place_medal: 
The fix works perfectly, unable to crash it anymore.
Thanks :)

Arne-W, 2018-03-17:
Nice to hear that we have found it :relaxed: Thank you for testing all my tries for fixing this issue :+1:

commit 81b79ff4 (2018-04-01): Graph View: Simplified dialog handling on file export. Fixes #1107

The QFileDialog used for selecting the export filename was implemented with a unnecessary signal/slot usage. It turned out that it was some sort of hack as it lead to a segfault on some systems.
closed 2018-04-01 by Arne-W