ArduPilot/apm_planner #1108: Logging does not work as expected

Arne-W, 2018-03-09:
On Ubuntu 17.10 it looks like the release define `QT_NO_DEBUG` forces the Qt logging system to suppress all log messages emitted with the `qDebug()` statement.
It seems like this is the normal behaviour - why this does not apply on Ubuntu 16.04 is still unclear.
Some information regarding this topic can be found in #1107 but all new information about the logging should be collected here.

[Some information about the Qt logging system]( http://www.kdab.com/wp-content/uploads/stories/slides/Day2/KaiKoehne_Qt%20Logging%20Framework%2016_9_0.pdf)

Arne-W, 2018-03-10:
I was able to reproduce this issue on Ubuntu 17.10. Until now I did not find a way to enable the debug logging in 17.10 even building a debug build did not do the trick.

drobban, 2018-03-10:
@Arne-W Look at my earlier comments in issue #1107 on how to enable logging in Ubuntu 17.10.

`qmake -config debug apm_planner.pro`
`make`
   (the newly built files are found in apm_planner/debug/apmplanner2

Add qtlogging.ini file inside ~/.config/QtProject/

My config looks like this;

> [Rules]
.debug=true
qt..debug=false

Walked this path myself and made it work, so take a look and let me know if that solved the problem.

Arne-W, 2018-03-11:
@drobban I thougt the `qtlogging.ini` would only work when using a category logger - but you are right using a `qtlogging.ini` did the trick also **without** using the `-config debug` switch. Thanks for remembering me! 
Moreover it also explains the different behavior on Ubuntu 16.04 and 17.10. On 17.10 you will find a default `qtlogging.ini` in `/etc/xdg/QtProject` which is missing in 16.04! This file looks like this:
> [Rules]
> .debug=false

Changing it like @drobban stated above enables the debug logging as expected. Only setting .debug to true foods the log with Qt internal logging.
Here you can find [more information](http://doc.qt.io/qt-5/qloggingcategory.html) about logging and the evaluating order of the different `qtlogging.ini` files
I think we should create an own category logger by default land use it with the `QLOG_DEBUG(), QLOG_INFO()...` macros - I have some ideas, will make a PR in the next days.

@AndKe Check the `qtlogging.ini` in `/etc/xdg/QtProject`

AndKe, 2018-03-11:
@Arne-W  *.debug=false here too :)

I'm glad to see you are working on a solution.

drobban, 2018-03-12:
@Arne-W glad that the .ini config solved the problem.
Reason why I insist on `--config debug` is that qDebug shouldn't work according to Qt-manual if QT_NO_DEBUG macro is defined. So to eliminate that as a potential factor... `--config debug`.

I might of read the manual wrong. But if I didn't read it wrong and you want to use the macro `NO_DEBUG`, then I would advice to change the macro name inside `logging.h` to use `QT_NO_DEBUG` instead & add config in the apm_planner.pro file to not include QT_NO_DEBUG.

As things are configured now, `QT_NO_DEBUG` define is added twice, once in apm_planner.pro and one more time by qmake after running `qmake apm_planner.pro`

drobban, 2018-03-12:
@Arne-W the manual I was talking about http://doc.qt.io/qt-5/qtglobal.html#qDebug

Arne-W, 2018-03-12:
@drobban As far as I can see the manual states that `QT_NO_DEBUG_OUTPUT` has to be defined at compile time to supress the `qdebug()` lgging. As this sort of define exists for Info and Warning level as well I would remove our own switches completely - only the trace will e left cause ist has no corresponding Qt level (is usese debug as well).  So if you want to switch off some loglevels at compile time you can use the QT defines for at least debug, info and warning. Moreover I will create a categorized logger for the whole planner and enable it for default by using `setFilterRules()` which overrides the content of the `qtlogging.ini`.
After that all users should see debug output again reagrdless what system type they are using. If anyone wants to change he / she could do it by specifying a `qtlogging.ini` with the `QT_LOGGING_CONF` environment variable.

AndKe, 2018-03-12:
with debug: 
```

[20180312 19:18:24.970 DEBUG] - write kml to  "/home/andre/Downloads/00-01-01_01-06-46.kml"
[20180312 19:18:24.974 DEBUG] - Done
[20180312 19:18:24.975 DEBUG] - "Successfull exported to /home/andre/Downloads/00-01-01_01-06-46.kmz."
[20180312 19:18:24.975 DEBUG] - KmlLogExporter::~KmlLogExporter()
[20180312 19:18:24.976 DEBUG] - LogExporterBase::~LogExporterBase()
[20180312 19:18:24.976 DEBUG] - Log export took  1458 ms

Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007ffff5d3372f in QDialog::accept() () from /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Widgets.so.5

```

commit f8a9e199 (2018-04-01): Logging: Created logging category and set it to debug level. Fixes #1108

Introduced an own logging category "apm.general" and set it to debug level to create the same logging output on all systems. Newer linux systems (like ubuntu 17.10) have switched of debug logging for Qt by default.
closed 2018-04-01 by Arne-W