ArduPilot/apm_planner #1109: libqt5serialport5 is installed, but qmake still gives me "Unknown module(s) in QT: serialport"

AniMysore74, 2018-03-11:
I get the following error when I try to build APM planner with qmake 

```

Project MESSAGE: Qt version 5.6.2
cat: /etc/redhat-release: No such file or directory
Project MESSAGE: Linux build x86
Project MESSAGE: Ubuntu Build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /home/aniruddha/Desktop/apm_planner DESTDIR /home/aniruddha/Desktop/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /home/aniruddha/Desktop/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
Project ERROR: Unknown module(s) in QT: serialport

```

I already have `libqt5serialport5` and `libqt5serialport5-dev` installed. I followed the official instructions for building on Linux. Also reinstalled both packages once, but no luck.

drobban, 2018-03-11:
Linux / dist version.
Version of apm_planner. what is the latest commit?
Are you able to run make?

AniMysore74, 2018-03-11:
Linux Version: Ubuntu 17.10.

I just cloned from the current master branch. 
Commit [9c606bb](https://github.com/ArduPilot/apm_planner/commit/9c606bb8682c79d20ae0ab72399d380336390331)

No, I'm not able to run make - no makefile has been generated by qmake.

drobban, 2018-03-11:
odd. done the procedure myself couple of times without problem.
Just performed all the steps without any problems.
My machine is running
```
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful

QMake version 3.1
Using Qt version 5.9.1 in /usr/lib/x86_64-linux-gnu
```

AniMysore74, 2018-03-12:
Fixed now. Seems anaconda's version of qt was interfering. 

```bash
ani@comp:~/Desktop/apm_planner$ qmake -v
QMake version 3.0
Using Qt version 5.6.2 in /home/aniruddha/anaconda3/lib
```

Temporarily commenting anaconda directory from `PATH` in `~/.bashrc`  and reloading bash works. Now system version of QT is loaded.

```bash
ani@comp:~/Desktop/apm_planner$ qmake -v 
QMake version 3.1
Using Qt version 5.9.1 in /usr/lib/x86_64-linux-gnu
 ```
 Built perfectly :+1:

closed 2018-03-12 by AniMysore74