ArduPilot/apm_planner #1110: APM Planner 2.5 for Mac Requesting Vehicle Firmware From Wrong URL

huttonsoftware, 2018-03-13:
I encountered a bug in APM Planner 2.5, and figured since I'm not to the point of being a developer I'd report it. When I went to Install Firmware versions, I was able to connect to the Pixhawk, but no icons for various vehicle frames appeared. I checked the Debug Output, and the program is using http://firmware.eu.ardupilot.org, which is apparently out of date. I'm running Mac OSX 10.12.6 Sierra. APM Planner 2.4 worked okay.  Here is the log:

[20180312 22:19:05.018 DEBUG] - ApmFirmwareConfig::requestFirmwares - Requesting firmware versions: "px4-v2"   "stable"
[20180312 22:19:05.448 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-heli/git-version.txt - server replied: Not Found"
[20180312 22:19:05.449 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.449 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Plane/stable/PX4/git-version.txt - server replied: Not Found"
[20180312 22:19:05.450 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.450 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-hexa/git-version.txt - server replied: Not Found"
[20180312 22:19:05.451 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.451 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Rover/stable/PX4/git-version.txt - server replied: Not Found"
[20180312 22:19:05.452 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.452 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-octa/git-version.txt - server replied: Not Found"
[20180312 22:19:05.453 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.453 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-quad/git-version.txt - server replied: Not Found"
[20180312 22:19:05.454 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.744 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-octa-quad/git-version.txt - server replied: Not Found"
[20180312 22:19:05.745 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.745 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4/git-version.txt - server replied: Not Found"
[20180312 22:19:05.746 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.747 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-tri/git-version.txt - server replied: Not Found"
[20180312 22:19:05.747 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)
[20180312 22:19:05.748 WARN ] - ApmFirmwareConfig:  "Error transferring http://firmware.eu.ardupilot.org/Copter/stable/PX4-y6/git-version.txt - server replied: Not Found"
[20180312 22:19:05.748 DEBUG] - firmwareListFinished error:  QNetworkReply::NetworkError(ContentNotFoundError)

drobban, 2018-03-13:
Not sure what the problem with the mirror site is.
but firmware is confirmed to be found here:
`http://firmware.ardupilot.org/Copter/stable/PX4-tri/`
`http://firmware.ardupilot.org/Copter/stable/PX4-tri/git-version.txt`

The mirror site doesn't seem to host the same content as firmware.ardupilot.org

Arne-W, 2018-03-13:
As far as I understand now the URL to search for new firmware should be `http://firmware.ardupilot.org` ?!
I can not remember why we changed it but it should be easy to fix as it is only defined here: [https://github.com/ArduPilot/apm_planner/blob/master/src/ui/configuration/ApmFirmwareConfig.cc#L38](https://github.com/ArduPilot/apm_planner/blob/master/src/ui/configuration/ApmFirmwareConfig.cc#L38)

drobban, 2018-03-13:
perhaps make it configurable?

Arne-W, 2018-03-13:
good Idea - would be "user fixable" then... I will have to do some tests whether its only the URL or if there are other issues.

drobban, 2018-03-13:
Asked on the ardupilot gitter for more info regarding the firmware site.
The info I got so far is that firmware.ardupilot.org is the one to be used according to a source in that channel @peterbarker I think it was & a recommendation to just use the px4 release, as all types are built into that release (if I understood correctly) according to @rmackay9. Also asked if there exist any released public statements about the firmwares that could be of interest. Haven't checked for replies on that, can them here if there are any

Arne-W, 2018-03-14:
I have seen the discussion. First we should use the `firmware.ardupilot.org` URL for a fast fix. Second step we should use the manifest.json file for getting the URL but I have not checked it until now.
The "all types in one firmware" thingy should have been already fixed last year - but I have to check that again...

billbonney, 2018-03-26:
@Arne-W We should push a new rc candidate for this version

Arne-W, 2018-03-26:
@billbonney but not without pulling #1113 in, as it fixes several other issues - especially a segfault happening on ubuntu 17.10 (see #1107)