ArduPilot/apm_planner #1119: Mac: "syntax error Unable to execute statement" while loading log

AndKe, 2018-04-18:
A friend experienced this using 2.0.24 on mac:  (while opening an arduplane dataflash log.)

[image]

most likely related to some mac library:

```
andre@Aranand:~/prog/apm_planner$ grep -inr "syntax error"
Binary file libs/lib/mac64/lib/libosgDB.dylib matches
Binary file libs/lib/mac64/lib/osgPlugins-3.0.1/osgdb_cfg.so matches
Binary file libs/lib/mac32/lib/libosgDB.dylib matches
Binary file libs/lib/mac32/lib/libcurl.4.dylib matches
Binary file libs/lib/mac32/lib/libexpat.a matches
Binary file libs/lib/mac32/lib/osgPlugins-3.0.1/osgdb_cfg.so matches
Binary file libs/lib/mac32/lib/libexpat.1.5.2.dylib matches
Binary file libs/lib/mac32/lib/libcurl.a matches
Binary file avrdude/avrdude.exe matches

```

Arne-W, 2018-04-19:
why is he using 2.0.24?? He should switch to a newer version - if the problem persists I am sure @billbonney will have a look.

AndKe, 2018-04-19:
that was my first advice :)  - I'll check if he did, and got it working.

AndKe, 2018-04-20:
current stable version works as expected.

closed 2018-04-20 by AndKe