ArduPilot/apm_planner #1121: Startup delay after use with Solo

AndKe, 2018-04-27:
please see this delay while waiting for Solo...  (there is no trace of that in GUI/connections)

```
[20180427 23:04:19.508 DEBUG] - APMToolBar setConnection: false
[20180427 23:04:19.508 DEBUG] - APMToolBar: updateLinkDisplay "No Devices" : "115200"  connected: false
[20180427 23:04:19.509 INFO ] - UDP Created  "UDP Link (port:14550)"
[20180427 23:04:19.509 INFO ] - UDPLink::UDP connect  QHostAddress(QHostAddress::Any) : 14550
[20180427 23:04:19.510 INFO ] - UDP: ADDING HOST: "127.0.0.1:46650"
[20180427 23:04:19.510 DEBUG] - HOST:  "127.0.0.1"
[20180427 23:04:19.511 DEBUG] - Address: "127.0.0.1"
[20180427 23:04:19.522 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180427 23:04:19.522 DEBUG] - HOST:  "10.1.1.1"
[20180427 23:04:19.548 DEBUG] - TryCorrectGoogleVersions, VersionGoogleSatellite:  "798"
[20180427 23:04:24.527 DEBUG] - Address: "10.1.1.1"
[20180427 23:04:24.540 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180427 23:04:24.541 DEBUG] - HOST:  "10.1.1.1"
[20180427 23:04:29.545 DEBUG] - Address: "10.1.1.1"
[20180427 23:04:29.559 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180427 23:04:29.560 DEBUG] - HOST:  "10.1.1.1"
[20180427 23:04:34.562 DEBUG] - Address: "10.1.1.1"
[20180427 23:04:34.610 DEBUG] - SCW: Link is disconnected
[20180427 23:04:34.627 INFO ] - No Ports Found ()
[20180427 23:04:34.628 DEBUG] - Baud rate: 115200 Expected: 4

```

the .ini file remembers it all to well..
```

[LINKMANAGER]
LINKS\1\baud=115200
LINKS\1\linkid=1
LINKS\1\port=No Devices
LINKS\1\type=SERIAL_LINK
LINKS\2\HOSTS\1\host=127.0.0.1
LINKS\2\HOSTS\1\port=46650
LINKS\2\HOSTS\2\host=10.1.1.1
LINKS\2\HOSTS\2\port=57060
LINKS\2\HOSTS\3\host=10.1.1.1
LINKS\2\HOSTS\3\port=57060
LINKS\2\HOSTS\4\host=10.1.1.1
LINKS\2\HOSTS\4\port=57060
LINKS\2\HOSTS\size=4
LINKS\2\linkid=2
LINKS\2\port=14550
LINKS\2\type=UDP_LINK
LINKS\size=2
LOGGING=true
PORTBAUDPAIRS\size=0

```

AndKe, 2018-04-30:
So the problem here is that when Solo , (or SITL) runs, it creates pointless entries \HOSTS\1...4
"Pointless" because they all send to UDP:14550 and are detected anyway.

On application start, all non-localhost connections then spend 15seconds with black screen..
I've made a very simple hack: 

https://github.com/AndKe/apm_planner/blob/b543c7610b363e0fa81ee8ad3585f05c56cd4ff8/src/comm/LinkManager.cc#L149-L156

it should maybe rather look for "." , working for any IP ?
or: even better - we should prevent those from be saved to the .ini in the first place ?

What do you think ?

Arne-W, 2018-05-02:
I think the solution should be preventing those from being written into the .ini file!
If it is reproduceable with SITL it should be possible to fix it without having a Solo. Until then it seems to be a workaround to remove all useless entires in the .ini file?!

AndKe, 2018-05-02:
yes, removing it from .ini is ok workaround.
I just happen to be modifying a bunch of solos these days, so I experience it more than usual, and "pollute" the .ini daily.  I am afraid other users may not "know" there is something wrong with .ini if they have only a slight delay after few IP connections.
- this is certainly not limited to solos only, it affects any mavlink over IP.

the localhost does not cause a delay, but I take if for granted that this line is caused by SITL connection:LINKS\2\HOSTS\1\host=127.0.0.1

Arne-W, 2018-05-16:
At the moment I do not understand what I have to do to add several HOST entries to a single UDP connection? Moreover I do not understand which use case depends on having more than one IP for a single connection?

AndKe, 2018-05-17:
I did nothing special to "achieve" that many host entries.
My general suggestion would be:
1-do not store autoconnected connection (if anything hits the open UDP port specified in connections , do not add it, add only manually entered IP connections)
2-clean up stored IP connections when updating to 2.0.26 (removing delays people already have)

Arne-W, 2018-05-17:
@AndKe Yes (1) could solve the problem, and (2) is always done if a new version is detected - the old ini is renamed and a new one is created.
But at the moment I try to understand why we need the possibility of having more than one IP adress for a connection - meaning why can we have more than one HOST entry for a LINK? Perhaps @billbonney can explain which use case is covered by this feature?!

Arne-W, 2018-05-17:
Ok I think I understand it now. It is used to handle multiple UAVs at the same time with one APM Planner - interesting use case but I think not used by the masses :wink: 
Nevertheless I do **not** have any delay on Kubuntu 16.04 when adding several hosts:
```[20180517 21:01:40.170 DEBUG] - Create Serial Connection: SerialConnection(0x37a74e0)
[20180517 21:01:40.171 INFO ] - "ttyS0" 115200
[20180517 21:01:40.305 DEBUG] - TryCorrectGoogleVersions, VersionGoogleSatellite:  "800"
[20180517 21:01:40.337 DEBUG] - APMToolBar: new Serial Link Created 1
[20180517 21:01:40.338 DEBUG] - APMToolBar setConnection: false
[20180517 21:01:40.338 DEBUG] - APMToolBar: updateLinkDisplay "ttyS0" : "115200"  connected: false
[20180517 21:01:40.339 INFO ] - UDP Created  "UDP Link (port:14550)"
[20180517 21:01:40.339 INFO ] - UDPLink::UDP connect  QHostAddress(QHostAddress::Any) : 14550
[20180517 21:01:40.340 INFO ] - UDP: ADDING HOST: "10.1.2.3:14555"
[20180517 21:01:40.340 DEBUG] - HOST:  "10.1.2.3"
[20180517 21:01:40.395 DEBUG] - Address: "10.1.2.3"
[20180517 21:01:40.396 INFO ] - UDP: ADDING HOST: "10.1.1.1:14555"
[20180517 21:01:40.396 DEBUG] - HOST:  "10.1.1.1"
[20180517 21:01:40.398 DEBUG] - Address: "10.1.1.1"
[20180517 21:01:40.398 INFO ] - UDP: ADDING HOST: "10.2.2.2:14123"
[20180517 21:01:40.399 DEBUG] - HOST:  "10.2.2.2"
[20180517 21:01:40.460 DEBUG] - Address: "10.2.2.2"
[20180517 21:01:40.461 INFO ] - UDP: ADDING HOST: "192.168.75.240:14555"
[20180517 21:01:40.461 DEBUG] - HOST:  "192.168.75.240"
[20180517 21:01:40.463 DEBUG] - Address: "192.168.75.240"
[20180517 21:01:40.464 INFO ] - UDP: ADDING HOST: "192.168.75.241:14555"
[20180517 21:01:40.464 DEBUG] - HOST:  "192.168.75.241"
[20180517 21:01:40.466 DEBUG] - Address: "192.168.75.241"
[20180517 21:01:40.466 INFO ] - UDP: ADDING HOST: "192.168.75.242:14555"
[20180517 21:01:40.466 DEBUG] - HOST:  "192.168.75.242"
[20180517 21:01:40.468 DEBUG] - Address: "192.168.75.242"
[20180517 21:01:40.479 DEBUG] - SCW: Link is disconnected
[20180517 21:01:40.488 DEBUG] - Baud rate: 115200 Expected: 4
```
and my ini looks the same:
```[LINKMANAGER]
LINKS\1\baud=115200
LINKS\1\linkid=1
LINKS\1\port=ttyS0
LINKS\1\type=SERIAL_LINK
LINKS\2\HOSTS\1\host=10.1.2.3
LINKS\2\HOSTS\1\port=14555
LINKS\2\HOSTS\2\host=10.1.1.1
LINKS\2\HOSTS\2\port=14555
LINKS\2\HOSTS\3\host=10.2.2.2
LINKS\2\HOSTS\3\port=14123
LINKS\2\HOSTS\4\host=192.168.75.240
LINKS\2\HOSTS\4\port=14555
LINKS\2\HOSTS\5\host=192.168.75.241
LINKS\2\HOSTS\5\port=14555
LINKS\2\HOSTS\6\host=192.168.75.242
LINKS\2\HOSTS\6\port=14555
LINKS\2\HOSTS\size=6
LINKS\2\linkid=2
LINKS\2\port=14550
LINKS\2\type=UDP_LINK
LINKS\size=2
LOGGING=true
PORTBAUDPAIRS\size=0
```
even when using the settings mentioned above:
```[20180517 21:16:29.325 DEBUG] - APMToolBar: updateLinkDisplay "No Devices" : "115200"  connected: false
[20180517 21:16:29.325 INFO ] - UDP Created  "UDP Link (port:14550)"
[20180517 21:16:29.325 INFO ] - UDPLink::UDP connect  QHostAddress(QHostAddress::Any) : 14550
[20180517 21:16:29.326 INFO ] - UDP: ADDING HOST: "127.0.0.1:46650"
[20180517 21:16:29.326 DEBUG] - HOST:  "127.0.0.1"
[20180517 21:16:29.326 DEBUG] - Address: "127.0.0.1"
[20180517 21:16:29.327 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180517 21:16:29.327 DEBUG] - HOST:  "10.1.1.1"
[20180517 21:16:29.328 DEBUG] - Address: "10.1.1.1"
[20180517 21:16:29.328 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180517 21:16:29.328 DEBUG] - HOST:  "10.1.1.1"
[20180517 21:16:29.329 DEBUG] - Address: "10.1.1.1"
[20180517 21:16:29.329 INFO ] - UDP: ADDING HOST: "10.1.1.1:57060"
[20180517 21:16:29.329 DEBUG] - HOST:  "10.1.1.1"
[20180517 21:16:29.330 DEBUG] - Address: "10.1.1.1"
[20180517 21:16:29.342 DEBUG] - SCW: Link is disconnected
[20180517 21:16:29.350 DEBUG] - Baud rate: 115200 Expected: 4
[20180517 21:16:29.350 DEBUG] - APMToolBar setConnection: false
```
It does not even take a second.
**BUT** as there is no way for the user to get rid of the unused Hosts except manipulating the .ini file 
I made a PR #1126 - @AndKe you should test and comment, also @billbonney should look!!

AndKe, 2018-05-17:
@Arne-W I think the time it takes is more about your routing table, whatever you get an instant "host unreachable" or there is an interface that have been used to reach it.
I'll try to test it within few days, (need first to reproduce the problem on current master and this computer, restore that .ini file )

Arne-W, 2018-05-18:
@AndKe So the issue is perhaps more a routing problem on host side than in APM planner? As you can see I tried with 192.168.75.xxx Adresses which are unknown hosts in my local lan and even that did not force the delay.

AndKe, 2018-05-18:
I cannot reproduce this on my home workstation, absolutely no delay.
The originally affected computer had a few wifi-cards (for RPAS comms) and were connected to internet using ethernet.
It's possible it have something todo with the routing there, (but it's only automatic/simple routing) in such case, there should be a limited amount of more advanced users that can see this delay, and not storing such connections multiple times can be a  n adequate solution.

Arne-W, 2018-05-22:
Not storing the same connection multiple times seems to be the right way. Creating a fix should not be that complicated.
I will close the #1126 PR.

Arne-W, 2018-08-08:
Made a new PR with a good fix - see #1142

closed 2018-08-12 by Arne-W

AndKe, 2018-10-22:
somehow this does not work anymore/or never worked , any colo connection from my laptop delays next AP2 start, and the settings looks like this only two Solo connections after deleting the .ini:
```
[LINKMANAGER]
LINKS\1\baud=115200
LINKS\1\linkid=1
LINKS\1\port=No Devices
LINKS\1\type=SERIAL_LINK
LINKS\2\HOSTS\1\host=10.1.1.1
LINKS\2\HOSTS\1\port=41677
LINKS\2\HOSTS\2\host=10.1.1.1
LINKS\2\HOSTS\2\port=39809
LINKS\2\HOSTS\size=2
LINKS\2\linkid=2
LINKS\2\port=14550
LINKS\2\type=UDP_LINK
LINKS\size=2
LOGGING=true
PORTBAUDPAIRS\size=0

```

Arne-W, 2018-10-23:
The connections have different ports. The fix only works for identical connections - meaning same ip AND same port.
I can change that to ip check only - but that would mean you can only store on connection per ip...

AndKe, 2018-10-23:
Even if only one connection is stored, it will delay the startup.
These connections are made without any user interaction, user starts AP2, Solo is connected right away, automatically.  So there is no value in saving them at all.
I suggest my early fix, - if IP is 10.1.1.1 , then do not store it.

Arne-W, 2018-10-29:
Ok - I will add a storage filter for 10.1.1.1! 👍

commit 58518c5d (2018-08-12): Comms: Linkmanager will store each UDP connection only once. Fixes #1121

and some cleanup...