ArduPilot/apm_planner #1127: UDPclient Not Connecting Properly

mjs513, 2018-05-26:
I am new to apm planner and mavlink so I am not 100% sure this is an issue or just my understanding of how it works.

On the communications drop down I selected add new link, UDP, and identified the IP:14550 as the connection.  Then I added a UDPclient link for IP:14555.  In both cases the IP address is the same, 192.168.1.162.  At the same time I clicked connect button.  Then closed the window.  

Right after I clicked on connect for UDP (14550) mavlink messages started streaming to APM Planner no issue.   However, when looking at my application I am not getting any messages from the UDPclient on 14555.  I ran Wireshark to see the traffic and noticed that the message that was suppose to be going to port 14555 was actually going to 14550.

In looking at other UDP issues I don't know if this is related or the result of a change that was made.

For reference I am using a Windows 10 64 bit laptop.  Not using a drone but using a Arduino linked to a ESP8266 to transfer messages to APM Planner.

Any help would be appreciated.

Thanks
Mike

Arne-W, 2018-05-28:
Hi @mjs513,
which Version aof APM Planner are you using??
Did I get you right that you are trying to send UDP 2 different ports at the same time?
Just for info - there should be an configuration folder for the planner somewhere in `c:\users\AppData`. I think it is called `apmPlanner2`. Inside this folder you will find an *.ini file having a part called `[LINKMANAGER]` whrere you can see all `HOSTS` and their port which are associated with a link. You can edit this directly and if you delete it it should be recreated on next start.
Moreover you will find some logfiles in that folder which you could analyze - searching for UDP should be a good start.

Arne-W, 2018-05-29:
You should use 2.0.26-rc2 as the version has a lot of bugs fixed... As the version is brand new you can find it only [here](https://www.dropbox.com/sh/pqd66ncnqw7km2b/AAAyCbYSaqanXIhckVNOeIzra?dl=0). It will be copied to the standart download location soon.

mjs513, 2018-05-29:
@Arne-W .  Thanks for the link will give it a try and see how it works.

mjs513, 2018-06-05:
@Arne-W..  Sorry for not getting back to you sooner on this.  For now I gave up on my approach and switched to using Mavesp8266 interfaced to the Teensy 3.5.  I did download and install the 2.0.26-rc2 and it is working great.  

Working through Mavlink and protocol is driving me nuts :).  But making a lot of progress and everything is working except for setting parameters from the GCS.  Think this is an issue with the Maveesp though, to be honest if I could compile ardurover from within the Arduino IDE and put it on a Teensy I would.  

Going to go ahead and close this.

Thanks for your help
Mike

closed 2018-06-05 by mjs513