ArduPilot/apm_planner #1130: Need MAVLINK 2 support.

AndKe, 2018-06-16:
This is getting increasingly important as mavlink2 is default for current ArduPilot builds.

billbonney, 2018-06-19:
I've never had time to analyze the impact of moving the MAVLINK2 in APM Planner. Best to look at https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc and compare with APM Planner 2 implementation.

The biggest challenge is probably not the integration of ML2 but the finding of the all the places that the code breaks.

AndKe, 2018-06-19:
"all other" GCS's handle mavlink2 now, so I guess you could ask other GCS devs about their experience.
Two weeks ago, one of the companies I helped came back to me and declared that AP2 does not work with Pixhawk2.1  - took me several minutes to figure out they loaded current AC from scratch, and had mavlink2.
It's a pity if others jump to this "conclusion".

billbonney, 2018-06-19:
Indeed, There is also Tower, and dronekit-android that will be lost by these changes to ue Mavlink2 exclusively. But we need someone or I need some free time to work on it.

Arne-W, 2018-06-19:
As the unit feature I am currently working on is nearly finished, I can have a look at this - hopefully I will find enough time.
@AndKe The current ArduCopter version is AFAIK 3.5.5 - is this version only supporting Mavlink2? Last version I have tested was 3.5.0 rc4 and everything was ok!

AndKe, 2018-06-19:
no, 3.5.5 can do mavlink(1.0) too, it's just that fresh installations default to 2.0 , I am not sure exactly when 2.0 became default.

Arne-W, 2018-06-19:
I do not get it - in the mavlink repository there is no released version called 2.0 - highest version  number is 1.0.11. Also the release notes of ArduCopter do not mention anything called Mavlink 2.0??
Could it be possible that the upcoming ArduCopter 3.6.0 will support Mavlink 2.0 ?? Since 1st of June rc2 is available - 3.6.0 is not official released until now.
Moreover the update to Mavlink 2.0 of Qgroundcontrol is 4 hours ago.... Now I am somehow confused....

AndKe, 2018-06-19:
no, I have one of their logs, it's 3.5.5 with
SERIAL0_PROTOCOL,2
SERIAL1_PROTOCOL,2

(I just assumed it were default, I doubt they fooled around with it)

billbonney, 2018-06-19:
see https://mavlink.io/en/guide/mavlink_2.html

Arne-W, 2018-06-19:
hmmm - on [https://mavlink.io/en/guide/mavlink_version.html](https://mavlink.io/en/guide/mavlink_version.html) they say version 2 is used since 2017... I will update my copter version to 3.5.5 and try to migrate to version 2 in the next week....perhaps it is not that difficult....

billbonney, 2018-06-19:
MAVLink 2 has been around awhile, just not mandatory til now

Arne-W, 2018-06-23:
@AndKe I did some testing yesterday and arducopter 3.5.5 supports Mavlink 1. I updated my copter to this version and everything works as expected. Moreover I tried to migrate to mavlink 2 library but Arducopter 3.5.5 only communicated with Mavlink 1 and due to the new library some messages are interpreted in an other way than the old messages which does not look right. I will post a screenshot of the mavlink inspector later.

AndKe, 2018-06-23:
Good to see you're in it.
I think you may have seen fallback to 1.0 , in any case, the easiest way would be to run AC3.6 RC in SITL. - which I think will default to Mavlink 2

AndKe, 2018-06-23:
@Arne-W The parameter question you asked on gitter.im is described in this thread 4 days ago :)
As for Mavlink1.0 , I don't think it will be removed any time soon, but Mavlink2 will soon be the default of all new installations (where the Ardupilot is not upgraded on a vehicle, but installed for the first time.)

Arne-W, 2018-06-24:
@AndKe Yes you are right - I did not understand everything from the beginning but now I got I think. Also that updating my copter did not show the mavlink change. I think I have to make SITL work first to be able to test.

Arne-W, 2018-06-24:
I thought I understood - but now totally confused.
I am using current SITL 3.6 dev with defaults and it connects like always to APM-Planner - I do not get the point of what has to be done or what is not working. Perhaps my SITL is communication by using Mavlink 1 but I did not find an easy way to figure out which version is **really** used. The Mavlink inspector in planner shows this:
[image]
WTF - Why is it mavlink version 3? I don't get it!

AndKe, 2018-06-24:
yes, the mavlink_version is a pesky thing, I understand it as kind of as a "sub-version" , maybe 3= 2.3 and 2=2.2  as long as protocol is 2.
In any case, I've seen "2" and maybe "3" in Mavlink1.0 and possibly 0.9

I am attaching a log gathered by somebody for HoTT vario, you can play it back using mavplayback.py mavlink2.tlog  
[mavlink2.zip](https://github.com/ArduPilot/apm_planner/files/2130998/mavlink2.zip)

AndKe, 2018-06-24:
@Arne-W 
It seems to me like your build works quite well, 
I see some messages that are not decoded, but it seems to work (tested with the .tlog file I posted)
`
[20180624 19:15:33.199 INFO ] - Unable to decode message from system  1  with message id: 125
[20180624 19:15:33.200 INFO ] - Unable to decode message from system  1  with message id: 152
[20180624 19:15:33.254 INFO ] - Unable to decode message from system  1  with message id: 178
[20180624 19:15:33.254 INFO ] - Unable to decode message from system  1  with message id: 182
[20180624 19:15:33.303 INFO ] - Unable to decode message from system  1  with message id: 129
[20180624 19:15:33.353 INFO ] - Unable to decode message from system  1  with message id: 137
[20180624 19:15:33.354 INFO ] - Unable to decode message from system  1  with message id: 163
[20180624 19:15:33.354 INFO ] - Unable to decode message from system  1  with message id: 165
[20180624 19:15:33.355 INFO ] - Unable to decode message from system  1  with message id: 2
[20180624 19:15:33.355 INFO ] - Unable to decode message from system  1  with message id: 147
[20180624 19:15:33.355 INFO ] - Unable to decode message from system  1  with message id: 193
[20180624 19:15:33.400 INFO ] - Unable to decode message from system  1  with message id: 241
[20180624 19:15:33.402 INFO ] - Unable to decode message from system  1  with message id: 65

`

billbonney, 2018-06-24:
I'll try and look later, but make sure you build mavlink with the https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml definitions for the extra ardupilot specific dialect

AndKe, 2018-06-24:
@billbonney Not sure who that "be sure" was meant for, so to clarify: (Just in case it was not meant for @Arne-W and AP2 build, but referred to the .tlog or the errors (me))

- the "unable to decode" is generated by playing back the previously attached .tlog (which is from real hardware, not SITL running a stable release.)   on AP2 from Arne-W's Mavlink_2 branch.    
-current AP2 master would not even respond to that data.

billbonney, 2018-06-24:
it was for @Arne-W as it seems to me to indicate that the message was a ArduPilot variant

Arne-W, 2018-06-25:
@billbonney Ha - I just took the original sources for mavlink 2 and did **not** generate them with the ardupilotmega.xml definitions. I think therefore we have the misinterpretation of the messages. I will try to generate the library by using those definitions this week.

billbonney, 2018-06-25:
From my very old notes I have something like

```
python -m pymavlink.tools.mavgen --lang C --wire-protocol 1.0 message_definitions/v1.0/ardupilotmega.xml
```

Hope that helps

Arne-W, 2018-06-25:
Found [https://github.com/mavlink/mavlink-devguide/blob/master/en/getting_started/generate_source.md](https://github.com/mavlink/mavlink-devguide/blob/master/en/getting_started/generate_source.md) today. I hope using the [ArduPilot/mavlink](https://github.com/ArduPilot/mavlink) code and the docs above will create perfect results :smiley:

Arne-W, 2018-06-27:
The problem was not really the message definition - the problem were the changed data structures of mavlink2 and hard coded offsets which didn't work any more. if someone wants to test this is my [current development branch](https://github.com/Arne-W/apm_planner/tree/Mavlink_2). Looks good so far - Mavlink inspector works and graphing too. It would be cool if you could check all functions you normally use and inform me if they do not work.
One issue I already know is that the planner crashes if you close it without disconnecting. It is triggered by a singleton which lives longer than several objects it deals with - as always singletons are a mess when shutting down applications...

Arne-W, 2018-06-27:
The crashing on close is already fixed... but tlogs which are saved using mavlink2 can not be parsed due to a problem when saving the received data but I am working on that...

AndKe, 2018-06-28:
@I am soon leaving for vacation, so I don't have good opportunity to test very well, however - I did pull your current branch , and as far I can see (by playing back misc logs), It works very well !  :)

Arne-W, 2018-06-28:
@AndKe thank you for testing - I wish you a nice holiday and have a good time! There will be enough time for finding bugs when you are back :wink:
BTW - saved tlogs can be parsed again....

Arne-W, 2018-06-28:
@billbonney the decoding problems @AndKe mentioned int [this post](https://github.com/ArduPilot/apm_planner/issues/1130#issuecomment-399779518) can be ignored I think - the mavlink decoder should do the same just more generic?!

Arne-W, 2018-07-23:
Solved this issue by adding these commits:
284b0fdb737da405a21b49c857643b6401ddedeb and 736532f14d6dd429f860627783936cceaae971ec

closed 2018-07-23 by Arne-W