ArduPilot/apm_planner #1136: Serious communication issue.

AndKe, 2018-07-27:
Curent master fail to do the "refresh parameters" ,  or fetch parameters on start, however, it does display broadcasted parameter changes.
Also, accel calibration fail:  user never gets the text instructions what to do (text from AP)

building 2.0.26-rc2 solves the problems.

This is most likely to be related to Mavlink 2.0  https://github.com/ArduPilot/apm_planner/commit/736532f14d6dd429f860627783936cceaae971ec

```
[20180727 10:38:36.235 WARN ] - Timeout requesting waypoint count - retrying.
[20180727 10:38:37.654 DEBUG] - ApmSoftwareConfig::populateTimerTick() - No Param names from param manager. Sleeping for one second...
[20180727 10:38:38.249 WARN ] - Timeout requesting waypoint count - retrying.
[20180727 10:38:38.599 DEBUG] - Critical message: "MAVLink Protocol" "Detected radio still using MAVLink v1.0 on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware."
[20180727 10:38:38.600 DEBUG] - Protocol Status Message: "MAVLink Protocol" "Detected radio still using MAVLink v1.0 on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware."
[20180727 10:38:38.600 DEBUG] - Switching outbound to mavlink 1.0 due to incoming mavlink 1.0 packet: 0x55971e422870 2 1
[20180727 10:38:38.705 DEBUG] - ApmSoftwareConfig::populateTimerTick() - No Param names from param manager. Sleeping for one second...
[20180727 10:38:39.057 DEBUG] - src/uas/UAS.cc 2171 LOADING PARAM LIST
[20180727 10:38:39.105 DEBUG] - "retrieve versionobject from server: http://firmware.ardupilot.org/Tools/APMPlanner/apm_planner_version.json"
[20180727 10:38:39.369 DEBUG] - Downloading: 2850 / 2850
[20180727 10:38:39.369 DEBUG] - Downloading: 2850 / 2850
[20180727 10:38:39.369 DEBUG] - AutoUpdateCheck::httpFinished()
[20180727 10:38:39.755 DEBUG] - ApmSoftwareConfig::populateTimerTick() - No Param names from param manager. Sleeping for one second...
[20180727 10:38:40.253 WARN ] - Timeout requesting waypoint count - retrying.
[20180727 10:38:40.475 DEBUG] - src/uas/UAS.cc 2171 LOADING PARAM LIST
[20180727 10:38:40.753 DEBUG] - ApmSoftwareConfig::populateTimerTick() - No Param names from param manager. Sleeping for one second...
[20180727 10:38:41.108 DEBUG] - Param: "STAT_FLTTIME" :  QVariant(int, 0)
[20180727 10:38:41.110 DEBUG] - Param: "STAT_RUNTIME" :  QVariant(int, 2055)
[20180727 10:38:41.452 INFO ] - "Requested retransmission of #1"
[20180727 10:38:41.453 INFO ] - "Requested retransmission of #2"
[20180727 10:38:41.453 INFO ] - "Requested retransmission of #3"
[20180727 10:38:41.453 INFO ] - "Requested retransmission of #4"
[20180727 10:38:41.453 INFO ] - "Requested retransmission of #5"
[20180727 10:38:41.803 INFO ] - "Requested retransmission of #1"
```

Arne-W, 2018-07-27:
@AndKe Welcome back - hope you had a nice holiday :smiley: 
Looks like you tested with a radio link?! Look at this log lines:
```
[20180727 10:38:38.599 DEBUG] - Critical message: "MAVLink Protocol" "Detected radio still using MAVLink v1.0 on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware."
[20180727 10:38:38.600 DEBUG] - Protocol Status Message: "MAVLink Protocol" "Detected radio still using MAVLink v1.0 on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware."
[20180727 10:38:38.600 DEBUG] - Switching outbound to mavlink 1.0 due to incoming mavlink 1.0 packet: 0x55971e422870 2 1
```
Seems like the fall back to mavlink 1 does not work. Could you please test if the problem persist when using an USB cable?

AndKe, 2018-07-27:
Unable to reproduce on direct serial connection with 3.5.7.

The cube that make me report this issue runs AC3.5.5  and does only have SERIAL*_PROTOCOL 1

The "on a link with MAVLink v2.0 enabled. Please upgrade the radio firmware." did not make any sense to me , as I do not know how it decided 2.0 was "enabled"   

The link is provided by OpenSolo 3.0.0 - so it's rather updated, but still - the cube is only speaking MAVLINK1 as always.

As a fun fact: even while the parameters stop working, - the "normal" telemetry works fine.

Arne-W, 2018-07-27:
I cannot reproduce this behavior. Tried with SITL 3.6-dev and the link forced to mavlink 1.0 and my Copter using 3.5.5 and SERIAL0_PROTOCOL set to 1.
It just works.... Will try to reproduce using a radio link...

Arne-W, 2018-07-27:
Even with my old radio link with FW 1.7 everything works! Seems to be somehow related to the Solo radio link... Seems this issue needs some deeper analyzing - a mavlink expert would be cool - could give some hints where to look next...

Arne-W, 2018-08-02:
@AndKe It would be very cool if you could test #1137 with OpenSolo. Hopefully the issue is fixed.

AndKe, 2018-08-03:
Yes, you fixed it ! :)  , but something needs fine-tuning:
On application start and first connection, it takes few minutes to get the parameters due to: 

```
[20180803 8:22:15.076 DEBUG] - Param: "EK2_YAW_I_GATE" :  QVariant(int, 300)
[20180803 8:22:15.175 DEBUG] - Param: "EK2_TAU_OUTPUT" :  QVariant(int, 25)
[20180803 8:22:15.178 DEBUG] - Param: "EK2_MAGE_P_NSE" :  QVariant(double, 0.001)
[20180803 8:22:15.181 DEBUG] - Param: "EK2_MAGB_P_NSE" :  QVariant(double, 0.0001)
[20180803 8:22:15.327 DEBUG] - Param: "EK2_RNG_USE_HGT" :  QVariant(int, -1)
[20180803 8:22:15.697 INFO ] - "Requested retransmission of #491"
[20180803 8:22:15.698 INFO ] - "Requested retransmission of #492"
[20180803 8:22:15.698 INFO ] - "Requested retransmission of #493"
[20180803 8:22:15.699 INFO ] - "Requested retransmission of #494"
[20180803 8:22:15.700 INFO ] - "Requested retransmission of #495"
[20180803 8:22:15.830 DEBUG] - Param: "EK2_TERR_GRAD" :  QVariant(double, 0.1)
[20180803 8:22:15.928 DEBUG] - Param: "EK2_BCN_M_NSE" :  QVariant(double, 1)
[20180803 8:22:15.931 DEBUG] - Param: "EK2_BCN_I_GTE" :  QVariant(int, 500)
[20180803 8:22:16.027 DEBUG] - Param: "EK2_BCN_DELAY" :  QVariant(int, 50)
[20180803 8:22:16.032 DEBUG] - Param: "EK2_RNG_USE_SPD" :  QVariant(double, 2)
[20180803 8:22:16.397 INFO ] - "Requested retransmission of #496"
[20180803 8:22:16.398 INFO ] - "Requested retransmission of #497"
[20180803 8:22:16.398 INFO ] - "Requested retransmission of #498"
[20180803 8:22:16.399 INFO ] - "Requested retransmission of #499"
[20180803 8:22:16.399 INFO ] - "Requested retransmission of #500"
```

on any following parameter reload (clicking refresh)   - it's fast and smooth, no re-requests.
If I exit & start AP2,  it is slow again - until parameters are fetched the first time.

AndKe, 2018-08-03:
Perfection !  :)  
- it works great. Thank you.
this alone with Mavlink 2.0 support deserves another AP2 release.

Arne-W, 2018-08-05:
fixed with b679433ca435a16c8a237c1c7dc455d53e090431 and 3e4adab0ab8ed8c508579f882c97996ecea01493

closed 2018-08-05 by Arne-W