ArduPilot/apm_planner #1139: ArduPilot - flashing Chibios firmwares

lvale, 2018-08-04:
At the moment APM Planner is only aware of px4 type firmware, and ArduPilot is moving away from these. Currently the newer firmware builds (apj) can+t even be selected as a custom firmware.

Arne-W, 2018-08-07:
@lvale making *.apj files available is no problem but what protocol is used for uploading? Is it the same as used by the pixhawk? If it is the same I would make a fast hack so you can test with other hardware.

kikislater, 2018-08-07:
"It uses the same protocol as the old PX4 bootloader, but uses different USB vendors and device IDs."
https://github.com/mavlink/qgroundcontrol/issues/6766

Arne-W, 2018-08-07:
@kikislater Ok that sounds good - do you eventually know where I can find those vendor and device IDs?

Arne-W, 2018-08-07:
...and does it use the default STM dfu mode for flashing like it is used by betaflight?
And how do I kick a stm target into dfu.... there are still a lot of questions...

kikislater, 2018-08-07:
Sorry don't really know. I just flash my pixhawk with chibios under mission planner
May be refering to this 

https://github.com/ArduPilot/MissionPlanner/commit/0f31685669715d5c4dccb5bc84491058accc7656

And this
https://github.com/ArduPilot/MissionPlanner/issues/1814

I know how to flash omnibus F4 with dfu but not pixhawk ...

kikislater, 2018-08-07:
Seems more described here : https://github.com/ArduPilot/ardupilot/blob/master/Tools/bootloaders/README.txt

And here http://ardupilot.org/dev/docs/porting.html

Arne-W, 2018-08-08:
First thanks for the information @kikislater !
Ok I think I got it - all boards which shall be able to use ardupilot first need a specific bootloader wich has to be flashed somehow. If this bootloader is available on the target the chibios firmwares can be flashed the same way the old nuttX firmwares were flashed?!
Would be cool if anyone could confirm my understanding...

kikislater, 2018-08-08:
If I understand well and that's it with Omnibus f4 pro : once bootloader is modify you could change whatever  chibios firmware you want (copter, plane, rover ...). And it was the same when I flashed my pixhawk clone 2.4.8 : I was asked to upload chibios bootloader instead of nuttx then be able to run an ardupilot fw on it ... 
I did not flash my others originals pixhawk 2.4.6 at this time. May be tagging an ardupilot developer like @tridge could help but I was thinking that this project was in relationship with them ;-)

lvale, 2018-08-09:
@Arne-W We've been discussing this, and the core explanation is right from @tridge :

For all the ChibiOS builds (including fmuvX), the product string in the 
USB descriptor matches the directory name on our firmware server. So if 
you see the following USB descriptor: 

 Manufacturor: ArduPilot 
 Product: FooBar 

then you can find the latest plane stable firmware for that board here: 

  http://firmware.ardupilot.org/Plane/stable/FooBar/arduplane.apj 

If it is in the bootloader (and is running the ArduPilot bootloader), 
then you will get this USB descriptor: 

 Manufacturor: ArduPilot 
 Product: FooBar-BL 

The only current exception to that rule is the CubeBlack, which gets 
this: 

 Manufacturor: Hex/ProfiCNC 
 Product: CubeBlack 

or in bootloader: 

 Manufacturor: Hex/ProfiCNC 
 Product: CubeBlack-BL 

and firmware is: 

  http://firmware.ardupilot.org/Plane/stable/CubeBlack/arduplane.apj 

so if you treat a manufacturer string of "Hex/ProfiCNC" as equivalent to 
a manufacturor of ArduPilot then it will all work out. 

For USB IDs, there are only three USB VIDs in use at the moment: 

 - the old 3DR one 0x26AC for the old bootloader 
 - the ST on 0x0483 (for most ChibiOS boards) 
 - the Hex/ProfiCNC one 0x2DAE 

You will notice that there are multiple firmwares that suit a particular 
board. For example, an old Pixhawk1 could load any of fmuv2, fmuv3 or 
Pixhawk1. 

We're moving towards the more user recognisable names (like Pixhawk1), 
but are keeping the old ones for now so existing firmware loaders can 
use them. 

You can see the full list here: 

  http://firmware.ardupilot.org/Plane/stable/ 

If the user has updated their bootloader with the new bootloader then 
their bootloader will have the product string matching the firmware 
name (with "-BL" on the end). 

Long term we'd like to get users onto the "named" firmwares, as we think 
it is more intuitive when a user selects a firmware to choose "Pixracer" 
(for example) rather than fmuv4. Only a programmer really knows that 
their board is a fmuv4, but the end user will recognise "Pixracer". 

Also note that if you prefer to use the 8 bit board type in the 
bootloader protocol, then there is now a file containing all of them for 
px4 and ArduPilot here: 

  https://github.com/PX4/Bootloader/blob/master/board_types.txt

Arne-W, 2018-12-10:
I just pushed a change de4262112220876e288ce94ae7642bff757f00db which should make it possible to flash chibiOS firmware by using the custom firmware button in APM-Planner. Perhaps someone would like to test.

bhaskarsdose, 2020-01-18:
can we build latest ardupilot firmware without chibi os directories or dependencies or can I port to a new controller board having nuttx like given here but without chibios:

https://ardupilot.org/dev/docs/porting.html

closed 2021-04-03 by Arne-W