ArduPilot/apm_planner #1144: Problem in opening dataflash log

anbello, 2018-08-12:
When I open logs created with Copter 3.6 dev I see a Warning dialog box with this text:

> There were data errors / unreadable data in the log! The data is potentially corrupt and incorrect.
> Detected:
> 3824 data corruptions

and as details several lines like this:

> Logline 99599: Corrupt data element found when decoding CTUN data.
> Logline 99628: Corrupt data element found when decoding CTUN data.
> Logline 99653: Corrupt data element found when decoding CTUN data.
> Logline 99680: Corrupt data element found when decoding CTUN data.
> 
> There were 3824 errors during log parsing.

If I click OK on the dialog box then I can see the graph but I am not sure that there are all the data.

Arne-W, 2018-08-12:
@anbello could you please check if you can find something like this
`[20180812 11:32:16.155 WARN ] - Float resolves to hard NaN - This is a serious log error as data is corrupted. Graphing may not work as expected for data of type  "CTUN"`
in the planner log and could you please upload your log here?

Seems like some copter 3.6-dev logs are corrupt some other users have the same problem [See here](https://discuss.ardupilot.org/t/arducopter-3-6rc-ctun-parameter-not-recognized/31766)

anbello, 2018-08-12:
@Arne-W yes there are plenty of warn like that, I attach both apm log and one dataflash log

[apmlog_20180812.txt.zip](https://github.com/ArduPilot/apm_planner/files/2280885/apmlog_20180812.txt.zip)
[18-08-10_18-46-16.zip](https://github.com/ArduPilot/apm_planner/files/2280886/18-08-10_18-46-16.zip)

Arne-W, 2018-08-12:
All right - those warnings are caused by different values used to detect a special log value (hard / soft NAN). For now you can ignore it until we found a solution.
Moreover the log which could be found in the forum entry I posted [here](https://github.com/ArduPilot/apm_planner/issues/1144#issuecomment-412330933) showed a bug in the unit handling of the logparser.
When fixing the unit issue I found that there is something fishy with the unit data in your log - but I am still working on it...

Arne-W, 2018-08-12:
I made a new PR #1146 which fixes the issue I mentioned above. If you use this PR you will see that the unit data is somehow corrupted in your log. The other log (from the forum) has valid unit data and is rc7. Perhaps something bad happened during the last commits?!

anbello, 2018-08-12:
With PR #1146 merged I have this message:
There were data errors / unreadable data in the log! The data is potentially corrupt and incorrect.
Detected:
314 format corruptions
3824 data corruptions

in details some lines of this kind:
Logline 850: Corrupt data element found when decoding CTUN data.
Logline 875: Corrupt data element found when decoding CTUN data.
Logline 899: Corrupt data element found when decoding CTUN data.

and some lines of this kind:
Logline 99703: Unit or scaling error. No unit description for data type RALY found
Logline 99703: Unit or scaling error. No multiplier description for data type RALY found
Logline 99703: Unit or scaling error. Found unit ID 115 but no matching unit
Logline 99703: Unit or scaling error. Found unit ID 63 but no matching unit
Logline 99703: Unit or scaling error. Found unit ID 63 but no matching unit
Logline 99703: Unit or scaling error. Found unit ID 63 but no matching unit
Logline 99703: Unit or scaling error. Found unit ID 63 but no matching unit
Logline 99703: Unit or scaling error. Found unit ID 63 but no matching unit

anbello, 2018-08-13:
@Arne-W Doesn't seem a problem of 3.6-dev vs 3.6-rc7, I have another quad with a pixracer and ArduCopter V3.6-dev (31bfedaf) that writes logs that don't give the Unit or scaling errors. The log I posted above and that gives those errors is from a quad with omnibusf4pro FC. It seems a problem of target, omnibusf4pro vs fmuv4.

Arne-W, 2018-08-13:
Intresting - isn't the binary generated from the same code? Perhaps @OXINARF and or @peterbarker should know that unit and scaling does not work on the omnibusf4pro target?!

anbello, 2018-08-13:
I posted this issue on ArduPilot/Chibios Gitter https://gitter.im/ArduPilot/ChibiOS

peterbarker, 2018-08-14:
@anbello @Arne-W My guess is that there's some log corruption going on here.  My understanding is that the omnibusv4pro has had logging issues in the past...

Arne-W, 2018-08-14:
@peterbarker [in this post](https://github.com/ArduPilot/apm_planner/issues/1144#issuecomment-412333647) you will find a log. It has some unit information but the most of the unit data is missing. If you use the apm planner including the #1146 PR it should log all errors regarding unit and type infos.

peterbarker, 2018-08-14:
@Arne-W Thanks.

That log's really weird.

The log does seem to be entirely well-formed, but missing many, many messages.  The FMTU messages you're seeing are from Log_Write messages - they're emitted by general code rather than startup routines.  I'm unable to reproduce the behaviour in SITL - and @anbello mentions it works fine on PixRacer.  Very, very strange.  It's almost like the state machines responsible for sending the MULT and UNIT messages aren't running.... I'll play on hardware tomorrow and see if I can replicate by dropping the buffer size.

[image]

@anbello That image shows your SD card going out to lunch periodically. This is the chief reason we have a largish buffer size for DataFlash logging in ArduPilot - SD cards suck.  To be clear - `DSF.Dp` is the number of packets we've been asked to log but don't have room for, and DSF.Mn is the minimum number of free bytes in the buffer over a period of time.

sh83, 2018-08-14:
@peterbarker Thanks for looking into it. Seems like 4k buffer is too small. With latest fixes to dma mem allocation there is at least 30kb avaliable, so 16kb buffer could be used. 
@anbello could you retest with large buffer sizes?

anbello, 2018-08-14:
@peterbarker thanks for analyzing the log.
@sh83 I already tested with LOG_BITMASK=830 and LOG_FILE_BUFSIZE=8 and 4 and I continue to see "Unit or scaling errors" Now I will test with LOG_FILE_BUFSIZE=16 and I will report.

anbello, 2018-08-14:
I tested with LOG_FILE_BUFSIZE=1 and the when I open the log I don't see any Unit or scaling errors moreover in the variable tree (on the right on the apm_planner windows) I see the units.
I post the log, it is from a little indoor test flight.

[80-01-01_01-00-00_18.zip](https://github.com/ArduPilot/apm_planner/files/2287440/80-01-01_01-00-00_18.zip)

sh83, 2018-08-14:
@anbello, @peterbarker seems like my commit reduced buffer size was the cause of the issue. Sorry about that. I will do pr reverting it.

anbello, 2018-08-14:
Thanks a lot @sh83

peterbarker, 2018-08-14:
On Tue, 14 Aug 2018, sh83 wrote:

> @anbello, @peterbarker seems like my commit reduced buffer size was the cause of the issue. Sorry about that. I will do pr reverting it.

@sh83 I don't think reverting that patch is the correct fix.  Reverting it 
would only hide a bug, I think.

There must be a bug here somewhere, as those messages are sent out as part 
of a state machine which should retry sending messages if there's no 
buffer space to send them.

sh83, 2018-08-14:
@peterbarker Ok, it would be better if dataflash could work with smaller buffers.

peterbarker, 2018-08-15:
On Tue, 14 Aug 2018, sh83 wrote:

> @peterbarker Ok, it would be better if dataflash could work with smaller buffers.

https://github.com/ArduPilot/ardupilot/pull/9195

OTOH - having larger buffers on these boards is only a good thing, as that 
graph showed!

Arne-W, 2018-08-20:
Some information for me - for remembering:
```
https://github.com/ArduPilot/pymavlink/blob/master/DFReader.py#L152
-> a hard-nan has zeroes in all of the trailing bits past the first 16
this value 0x7FCA9222 but should have 0x7FC04152...
The float soft NAN is 0x7FC04152 like expected the float hard NAN is 0x7FC00000
```

commit 718b06eb (2018-08-30): GraphView: Changed NaN detection Fixes #1144

NaN detection changed from detecting soft NaN and treat every other value as hard NaN to detecting hard NaN and treat all other as soft NaN.
closed 2018-08-30 by Arne-W