ArduPilot/apm_planner #1147: apmplanner2 requires version 5.9.0 or later, but QtOpenGL provides version 5.6.0

jolly99, 2018-08-13:
When I am trying to start APM 2.0.26 on MacOS 10.13.5 I am getting the following error:
Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtOpenGL.framework/Versions/5/QtOpenGL
  Referenced from: /Applications/APM Planner 2.0 2.app/Contents/MacOS/apmplanner2
  Reason: Incompatible library version: apmplanner2 requires version 5.9.0 or later, but QtOpenGL provides version 5.6.0

Binary Images:
       0x10ba2b000 -        0x10c5eefff +org.ardupilot.apmplanner2 (???) <471C336F-6BAF-32AE-B650-248C86B2BF28> /Applications/APM Planner 2.0 2.app/Contents/MacOS/apmplanner2
       0x10ca20000 -        0x10cae7fff +SDL2 (3.1) <78D9C537-E977-3AD1-9890-53FBC62A4D69> /Applications/APM Planner 2.0 2.app/Contents/Frameworks/SDL2.framework/Versions/A/SDL2
       0x10f20a000 -        0x10f2549df  dyld (551.3) <AFAB4EFA-7020-34B1-BBEF-0F26C6D3CA36> /usr/lib/dyld
    0x7fff3916c000 -     0x7fff3916cfff  com.apple.ApplicationServices (48 - 50) <7BD49390-6D89-3429-80CD-6C70334F9B49> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices
    0x7fff6295f000 -     0x7fff62971ffb  libz.1.dylib (70) <48C67CFC-940D-3857-8DAD-857774605352> /usr/lib/libz.1.dylib

Arne-W, 2018-08-14:
@billbonney Could you please have a look as I do not have a Mac to test the build?!

sauloonze, 2018-08-16:
I have the same issue.

mgcat, 2018-08-17:
Also have the same issue here. On my older Mac OS X 10.10 Air it did Ok but not on my new 10.13 MacBk Pro

alexhermida, 2018-08-20:
Same issue here! macOS High Sierra 10.13.6.

Arne-W, 2018-08-21:
Using the daily is a good workaround - our Mac expert is on vacation at the moment.
The daily with the newest date is always the master build. If you want to use the 2.0.26 you should use this daily build [http://firmware.ardupilot.org/Tools/APMPlanner/daily/2018-05-24/](http://firmware.ardupilot.org/Tools/APMPlanner/daily/2018-05-24/) it should be the same version.

billbonney, 2018-08-21:
you can use a newer daily build http://firmware.ardupilot.org/Tools/APMPlanner/daily/2018-08-18/ at that works as expected.

jeffstreet, 2018-09-05:
Thanks @billbonney. Has the same issue and the daily fixed it on MacOS 10.13.6.

zakkhoyt, 2019-08-16:
Had same issue as OP on macOS catalina beta 5. Daily build [2019-03-08](http://firmware.ardupilot.org/Tools/APMPlanner/daily/2019-03-08/ )
launches for me.

closed 2021-02-21 by Arne-W