ArduPilot/apm_planner #1152: Please add support for dataflash log review of latest ArduPilot Firmware

SwiftGust, 2018-09-10:
I really liked using APMplanner2 for reviewing dataflash log.
It gives great automatic scaling, coloring, event messages...
However, now it looks like APMplanner2 does not compatible with datalog of latest ArduPilot firmware
Please add support for latest dataflash log of ArduPilot 

[image]

Arne-W, 2018-09-10:
@SwiftGust please update to APM-Panner 2.0.26-rc2 which can be found [here](http://firmware.ardupilot.org/Tools/APMPlanner/beta/). It is the last released stable version.
If you want the newest stuff - should be very stable too - you could look in [in this forum post](https://discuss.ardupilot.org/t/arducopter-3-6rc-ctun-parameter-not-recognized/31766). The win 64 installer should be build from the current master.

SwiftGust, 2018-09-10:
Thank you for quick reply.
It works fine,
but has little problem with ArduCopter dataflash...

[image]

Arne-W, 2018-09-10:
@SwiftGust Yes its a known problem - see the Forum post - that version should fix it. Moreover the error message is no problem an can be ignored....

SwiftGust, 2018-09-10:
Yes, no problem at all for reviewing flight data. 
Thanks

closed 2018-09-10 by SwiftGust