ArduPilot/apm_planner #1153: Problem with Arducopter 3.6, MAVLink2 and Radio Calibration Page

anbello, 2018-09-16:
If I set SERIAL0_PROTOCOL = 2 (MAVLink2) then on Radio Calibration Page I see all values 0 and the graphics don't follows my sticks, in MAVLink Inspector I see the correct values.
With SERIAL0_PROTOCOL = 1 (MAVLink1) is all OK.

Arne-W, 2018-09-17:
@anbello 2.026-rc2 does not have support for MAVLink 2 - sorry. We introduced the MAVLink2 support right after the las release. If you want to use MAVLink2 you will have to take the current master.
If you are on Windows you could find a build of the master [in this forum post](https://discuss.ardupilot.org/t/arducopter-3-6rc-ctun-parameter-not-recognized/31766/8). If you are on linux I can try to build a version for you this evening and if you are on mac you sould look [here](http://firmware.ardupilot.org/Tools/APMPlanner/daily/2018-08-30/)

anbello, 2018-09-17:
The version I am using and for which I report this issue is built by me from master (commit 718b06e).
What I wrote under [EDIT] is the version reported from Help -> About.

Arne-W, 2018-09-17:
Ah - Ok (did not check the git hash) I will try to reproduce it in the naxt days....

Arne-W, 2018-09-18:
Could reproduce this issue yesterday. Looks like the MAVLink message ID has changed cause MAVLink 2 message supports 16 RC channels. I hope I can fix it and perhaps extend the gui to show all 16 channels on next weekend...

Arne-W, 2018-09-23:
@anbello If you like you can test [my branch](https://github.com/Arne-W/apm_planner/tree/rc_channels). It should fix this issue and show all 16 chnnels. It is still a WIP so if you have some Ideas for making it better - just let me know.

anbello, 2018-10-02:
I am sorry for so late replay. Unfortunately, in this period I have little time for these stuff.
Anyway I tested your branch and it works good for me.

PR #1158 (2018-10-05): Radio Setup: Added compatibility for MAVLink 2 and increased to 16 channels
Radio config dialog will now work again when using MAVLink 2.
Supports now up to 16 RC channels.

Fixes #1153
commit 98cf653e (2018-10-05): Radio Setup: Added compatibility for MAVLink 2 and increased to 16 channels. Fixes #1153

Radio config dialog will now work again when using MAVLink 2.
Supports now up to 16 RC channels.
closed 2018-10-05 by Arne-W