ArduPilot/apm_planner #1155: Graph: "Show Mode" does not show mode for loaded .tlog

AndKe, 2018-09-20:
it does, however, appear to work for graphing tlog data as they come in, but for logged graph

Arne-W, 2018-09-20:
Looking at the code I think it has never worked cause I used the wrong datafield for extracting the mode from the heartbeat message....
If you have some older logs where you can compare tlog and bin logs of the same flight it would be interesting if it has ever worked correctly?

Arne-W, 2018-09-21:
Thanks for testing - as you can see I made a fix.
Your last log should not have worked too - did the mode change and was it different from auto or guided?

AndKe, 2018-09-21:
now that you ask, I see there is something very strange with the "heli" finding.
It shows Guided and AltHold - but when I configure "table filter" for HEARTBEAT only, I can't find any of those changes in HEARTBEAT.custom_mode

Arne-W, 2018-09-21:
without the bugfix the mav type was interpreted as mode - you should see a change in data there.

AndKe, 2018-09-21:
yep, you are perfectly correct.  :)