ArduPilot/apm_planner #1157: Map: visualize guided mode target.

AndKe, 2018-09-26:
Once user clicked map to do a guided-mode command, it would be nice to place some symbol on the spot, to indicate the target.

Arne-W, 2018-09-26:
You are right that would be cool - shall it be also removed as soon as the UAS reaches that location??

AndKe, 2018-09-26:
No, I would suggest to keep it until mode were switched away from guided.
This way, anyone can see, at a glance, not only why the RPAS is circling "somewhere" , but "why" , in the sense that there is something visible on the map.
- Maybe it's possible to a mark and a string like "guided, 100m REL" beside that mark. ?

Another, very related suggestion would be: 
Guided is sometimes used for holding in a safe area due to other traffic in CTR, it requires right click already,  - so it would be nice to have the GUIDED mode to be activated as a part of the "right click and select" action.   (not require pilot to switch to guided, then right click.)