ArduPilot/apm_planner #1162: more than one MAVlink device confuse AP2

AndKe, 2018-12-08:
This video is current master + 3DR Solo (3.6.x firmware+ MAVlink gimbal) 
https://youtu.be/N226kRA93H4

Please observe how it jump around the parameter count, often displaying "50" as the total, then going far beyond the maximum possible parameters.

Arne-W, 2018-12-12:
@AndKe Multiple mavlink devices are not really supported I think - but if you have a log I will check if we can do something about that.

AndKe, 2018-12-12:
I guess all these logs are from Solo (you should spot problems easily by extreme amount of params being transmitted.)
[tlogs with solo.zip](https://github.com/ArduPilot/apm_planner/files/2672856/tlogs.with.solo.zip)

The biggest file, holds 1880  parameter -lines ..., the GMB_ parameters are related to the gimbal.

As for the "not really supported" , I can say that multiple vehicles works fine(not all GUI is updated when switching), but at least does not cause trouble.  This is same vehicle, but some parameters belong to another subsystem, they could very well be presented in the same full parameter view. 
No special handling/GUI needed, just do not let the parameter requests go bananas :)