ArduPilot/apm_planner #1166: Tried building on OS X: "No rule to make target `QOSGWidget.h'"

drnic, 2019-02-10:
Any pro tips on building for OS X?

```
$ brew install qt5
$ export PATH="/usr/local/opt/qt/bin:$PATH"
$ qmake apm_planner.pro
Project MESSAGE: Qt version 5.12.0
Project MESSAGE: Mac build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /Users/drnic/Projects/flightcontrollers/apm_planner DESTDIR /Users/drnic/Projects/flightcontrollers/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /Users/drnic/Projects/flightcontrollers/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View

$ make
make: *** No rule to make target `QOSGWidget.h', needed by `build-release/ui/ui_QMap3D.h'.  Stop.
```

Ralender, 2019-02-20:
same issue on Archlinux

> $ qmake apm_planner.pro
> Project MESSAGE: Qt version 5.12.1
> cat: /etc/redhat-release: No such file or directory
> Project MESSAGE: Linux build x86
> Project MESSAGE: Enabling flite audio support
> Project MESSAGE: Release flavor
> Project MESSAGE: BASEDIR /home/USER/tmp/apm_planner DESTDIR /home/USER/tmp/apm_planner/release TARGET apmplanner2
> Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
> Project MESSAGE: Skipping support for Google Earth
> Project MESSAGE: Skipping support for libfreenect
> Project MESSAGE: Adding ALGLIB /home/USER/tmp/apm_planner/libs/alglib
> Project MESSAGE: Skipping support for XBee
> Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
> Project MESSAGE: Skipping support for Opal-RT
> Project MESSAGE: Skipping support for Google Earth
> Project MESSAGE: Skipping support for Camera View
> $ make
> make: *** No rule to make target 'QOSGWidget.h', needed by 'build-release/ui/ui_QMap3D.h'.  Stop.

NikolaeVarius, 2019-03-06:
Same here on Manjaro

`qmake apm_planner.pro                                                                                                     ✘ 2 
Project MESSAGE: Qt version 5.12.1
cat: /etc/redhat-release: No such file or directory
Project MESSAGE: Linux build x86
Project MESSAGE: Enabling flite audio support
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /home/jkim/Projects/apm_planner DESTDIR /home/jkim/Projects/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /home/jkim/Projects/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
$ make
make: *** No rule to make target 'QOSGWidget.h', needed by 'build-release/ui/ui_QMap3D.h'.  Stop.
`

Arne-W, 2019-03-08:
Hi guys - sorry it took so long to get an answer. I am able to reproduce it - seems to be somehow Qt 5.12.X related, using Qt 5.9.7 everything works.
I will have a look and try to fix this issue

Arne-W, 2019-03-08:
Hope this fixes the issue :smiley: just try the master.
If it does not work let me know.

closed 2019-03-08 by Arne-W

NikolaeVarius, 2019-03-11:
Confirmed building. Thank you.

commit 299ff23b (2019-03-08): Removed QMap3d.ui as is was not used. Fixes #1166

Moreover it leads to a failing build when using Qt 5.12.X