ArduPilot/apm_planner #1169: Unable to start on archlinux

a-wing, 2019-04-02:
git: master 299ff23b5e9910de04edfc06b6893bb06b47a57b

Running has been here

OS: archlinux
qt: 5.12.2

```
[20190402 14:06:32.336 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:32.392 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555894e01b50)
[20190402 14:06:32.392 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:32.440 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555894e51f20)
[20190402 14:06:32.440 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:32.498 DEBUG] - Creating MainWindow
[20190402 14:06:35.504 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558952b9600)
[20190402 14:06:35.504 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:35.544 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558953c28d0)
[20190402 14:06:35.545 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:35.582 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55589540f460)
[20190402 14:06:35.582 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:35.628 DEBUG] - Creating MainWindow
[20190402 14:06:38.440 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558958868f0)
[20190402 14:06:38.440 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:38.496 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55589597c610)
[20190402 14:06:38.496 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:38.541 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558959cc480)
[20190402 14:06:38.541 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:38.598 DEBUG] - Creating MainWindow
[20190402 14:06:41.647 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555895e9d140)
[20190402 14:06:41.647 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:41.689 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555895f33e80)
[20190402 14:06:41.689 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:41.727 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555895f83f80)
[20190402 14:06:41.727 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:41.772 DEBUG] - Creating MainWindow
[20190402 14:06:45.453 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558963fda60)
[20190402 14:06:45.453 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:45.521 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558964ee8a0)
[20190402 14:06:45.521 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:45.575 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55589653b620)
[20190402 14:06:45.575 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:45.632 DEBUG] - Creating MainWindow
[20190402 14:06:48.391 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555896a31b20)
[20190402 14:06:48.391 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:48.425 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55588cfb69f0)
[20190402 14:06:48.425 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:48.457 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555896af4e40)
[20190402 14:06:48.457 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:48.492 DEBUG] - Creating MainWindow
[20190402 14:06:51.031 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555896f5dc50)
[20190402 14:06:51.031 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:51.066 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55589706f6f0)
[20190402 14:06:51.066 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:51.096 INFO ] - Creating SubMainWindow:  SubMainWindow(0x5558970bf590)
[20190402 14:06:51.096 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:51.130 DEBUG] - Creating MainWindow
[20190402 14:06:53.307 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55588b80dfd0)
[20190402 14:06:53.307 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:53.341 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555897633ae0)
[20190402 14:06:53.341 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:53.370 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555897683ba0)
[20190402 14:06:53.370 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:53.403 DEBUG] - Creating MainWindow
[20190402 14:06:55.676 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555897af2f50)
[20190402 14:06:55.676 INFO ] - setting objectName:  "VIEW_MISSION"
[20190402 14:06:55.710 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555897be84b0)
[20190402 14:06:55.711 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190402 14:06:55.744 INFO ] - Creating SubMainWindow:  SubMainWindow(0x555897c38420)
[20190402 14:06:55.744 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190402 14:06:55.790 DEBUG] - Creating MainWindow

```

Arne-W, 2019-04-28:
I am building and running on Ubuntu 18.04. with Qt 5.12.2 and I can not reproduce you issue.
The log should look like this:
```[20190428 14:28:04.593 INFO ] - QGCCore::initialize()
[20190428 14:28:04.593 INFO ] - Current Build Info
[20190428 14:28:04.594 INFO ] - Git Hash: db0a67b0752d6ad511c6c764f5e5f37ecebc2d1e
[20190428 14:28:04.594 INFO ] - Git Commit: 2.0.27-rc1-9-gdb0a67b07
[20190428 14:28:04.594 INFO ] - APPLICATION_NAME: "APM Planner"
[20190428 14:28:04.594 INFO ] - APPLICATION_VERSION: "v2.0.27-rc1"
[20190428 14:28:04.594 INFO ] - APP_PLATFORM: ubuntu64
[20190428 14:28:04.594 INFO ] - APP_TYPE: stable
[20190428 14:28:04.745 INFO ] - Start Link Manager
[20190428 14:28:04.745 DEBUG] - Create MAVLinkDecoder:  MAVLinkDecoder(0x55a775dda540)
[20190428 14:28:04.747 INFO ] - Start UAS Manager
[20190428 14:28:04.816 DEBUG] - Creating MainWindow
[20190428 14:28:04.995 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55a775e526b0)
[20190428 14:28:04.995 INFO ] - setting objectName:  "VIEW_MISSION"
[20190428 14:28:05.028 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55a775f5c620)
[20190428 14:28:05.028 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20190428 14:28:05.050 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55a775fcb170)
[20190428 14:28:05.050 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20190428 14:28:05.281 DEBUG] - fillPortsInfo 
[20190428 14:28:05.358 WARN ] - libpng warning: iCCP: known incorrect sRGB profile
[20190428 14:28:05.360 WARN ] - libpng warning: iCCP: known incorrect sRGB profile
[20190428 14:28:05.361 WARN ] - libpng warning: iCCP: known incorrect sRGB profile
[20190428 14:28:05.461 INFO ] - Creating SubMainWindow:  SubMainWindow(0x55a776058dc0)
[20190428 14:28:05.461 INFO ] - setting objectName:  "VIEW_SOFTWARE_CONFIG"
[20190428 14:28:05.469 DEBUG] - Disable Controls 
```

Arne-W, 2019-06-09:
@a-wing Is this still an Issue or did you find a solution??

fllmn, 2019-06-20:
I'm experiencing the same problems on arch with plasma 5.16.0 on 2.0.27-rc1-9-g80b62f1c7-DEV

I noticed it was starting when running under sudo but that's not really an option for me so i complied with debug flags and found the issue being ApmHardwareConfig. I've not found a solution but commenting out the "firmwareConfig" block at least starts the application.

```
    ui.antennaTrackerLargeButton->setVisible(false); // [HIDE Antenna Tracking]
/*
    m_apmFirmwareConfig = new ApmFirmwareConfig(this);
    connect(m_apmFirmwareConfig,SIGNAL(showBlankingScreen()),this,SLOT(activateBlankingScreen()));
    ui.stackedWidget->addWidget(m_apmFirmwareConfig); //Firmware placeholder.
    m_buttonToConfigWidgetMap[ui.firmwareButton] = m_apmFirmwareConfig;
    connect(ui.firmwareButton,SIGNAL(clicked()),this,SLOT(activateStackedWidget()));
    connect(this, SIGNAL(advancedModeChanged(bool)), m_apmFirmwareConfig, SLOT(advancedModeChanged(bool)));
*/
    m_flightConfig = new FlightModeConfig(this);
```