ArduPilot/apm_planner #1171: Blank Install Firmware Screen on latest build 2.0.27-rc1

stejesh, 2019-04-30:
Built apm planner 2.0.27-rc1 from source today on Ubuntu 16.04
This is my 'install firmware' screen which is blank in both disconnected & connected state (I'm trying to configure my pixhawk 2.1 for the first time)

Any ideas why this is occuring and how do I fix this?

[image]

Arne-W, 2019-05-01:
Hi @stejesh , could you please upload the logfile? There should be an apmplanner2 folder in your home directory. In this folder you should find some logfiles called apmlog_2019xxyy.txt. The Apm-Planner generates one logfile per day.
With the logfile we should be able to find the Problem.

Arne-W, 2019-05-03:
@stejesh Did you see my message above?? I can not do anything without the logfile...

technicalfiddles, 2020-01-17:
I have the same issue. It looks like APM Planner 2 is querying the wrong address to load firmware. I've attached my log file.

I have a Pixhawk4 and the following
https://firmware.ardupilot.org/Rover/stable/PX4/git-version.txt

[apmlog_20200117.txt](https://github.com/ArduPilot/apm_planner/files/4077278/apmlog_20200117.txt)

Arne-W, 2020-01-19:
@technicalfiddles Cool - thanks for the log. I will have a look

technicalfiddles, 2020-01-24:
Thanks Arne. Any suggestion on loading firmware in the meantime? Can I bypass those links somehow?

Arne-W, 2020-01-25:
@technicalfiddles Yes you can go to [https://firmware.ardupilot.org/](https://firmware.ardupilot.org/) and download the firmware for your target. In APM Planner go to Initial Setup -> Install Firmware and click Custom. You will be prompted to select a file - select the one you downloaded and it will be flashed to your target.

technicalfiddles, 2020-01-29:
Arne - unfortunately I don't see a Custom button. See below:

[image]

Arne-W, 2020-01-30:
@technicalfiddles Ah - sorry I forgot to say that you have to enable the "Advanced mode" in the file menu. I never use the planner without turning it on - so I forgot to tell you.

Andr3g2, 2020-05-08:
hallo  stejesh 

missing button, which suggests a missing something to make.
i have make 2.0.27-rc1 on ubuntu 16 successful.

check my prerequisite:
apt-get install git qt5-qmake qt5-default \
  qtscript5-dev libqt5webkit5-dev libqt5serialport5-dev \
  libqt5svg5-dev libsdl1.2-dev  libsndfile-dev \
  flite1-dev libssl-dev libudev-dev libsdl2-dev \
  qtdeclarative5-qtquick2-plugin

regards andre

closed 2021-04-03 by Arne-W

Arne-W, 2021-04-03:
You should test it first :smile:

AndKe, 2021-04-03:
Tested on CubeBlack + copter.
Worked fine.
Please change "flashing firmware" as the progress bar label to "Done" .. when done.
(people may focus on during flashing, and it is left with "flashing" while the process is done.)

I love the flexibility and lots of options the new method provides.

commit 5d3f27e7 (2021-04-03): Firmware Install: Completely rewritten and redesigned. Fixes #1171 #1139 #1111

Created a completely new firmware update dialog. I uses the JSON firmware description and should now work with every firmware and hardware which is able to flash apj or px4 files.