ArduPilot/apm_planner #1172: APM Planner 2.0 cannot be opened

deandrecapati, 2019-05-31:
I'm trying to install APM Planner for macOS Mojave, but it won't let me open the program. I tried reinstalling the program as suggested, but not luck. Anyone have any suggestions on the issue or know about the issue?

Arne-W, 2019-06-09:
@deandrecapati did you use the newest version? 
Please try this version: [http://firmware.ardupilot.org/Tools/APMPlanner/daily/2019-03-08/apm_planner_2.0.27-rc1-7-g299ff23_osx.dmg](http://firmware.ardupilot.org/Tools/APMPlanner/daily/2019-03-08/apm_planner_2.0.27-rc1-7-g299ff23_osx.dmg)

kelechiokwuriki, 2020-04-05:
My macos version is Catalina 10.15.3 but it still did not open when i tried to install

mboland, 2020-04-05:
Did you allow it to open in System Preferences/Security & Privacy/General/app Store and Identified Developers?

MonkeyManGuy, 2020-08-03:
> Did you allow it to open in System Preferences/Security & Privacy/General/app Store and Identified Developers?
> Works fine here on 10.15.4

All Mac users are accustomed to doing that.

Catalina 10.15.6 here and I get a message saying APM Planner is not compatible with this version of OSX. This is after a fresh install

mboland, 2020-08-03:
@MonkeyManGuy 
I am currently using 10.15.6 without issue.
The ‘not compatible’ message is usually for 32bit applications.
Does it have a circle with slash through it on the icon.

MonkeyManGuy, 2020-08-04:
Take a look at these screenshots:

billbonney, 2020-08-04:
Just download the latest release candidate and you will be golden! https://github.com/ArduPilot/apm_planner/releases

MonkeyManGuy, 2020-08-04:
This are the pictures I meant to get attached. These were taken on 7.31.2020 with the latest download. But I will uninstall it and try again with the latest download once more & report back.

[image]

[image]

MonkeyManGuy, 2020-08-04:
It happens again. Ive opened up the details this time and it shows something interesting:
[image]
[image]

mboland, 2020-08-04:
Firstly, the FTDI driver package IS NOT compatible.
As in its name on the drive 10.4 to 10.7
You should not need to install drivers on 10.15

Installing is just a matter of dragging APM into your Apps folder 
The message you get is a little different

To illustrate I did a fresh download from the Link @billbonney provided above 
1: just drag the APM-P into your app folder from the downloaded .dmg package
[image]
When you double click it you will get this message
[image]
Go to System Preferences/Security & Privacy and click ‘Open Anyway’ You may need to unlock the pane first.
[image]
When it opens you will get an additional warning
[image]
Click open and all is done.

drobban, 2020-08-07:
> I had been downloading the software from here in the 3 times Ive attempted it since posting here: https://firmware.ardupilot.org/Tools/APMPlanner/
> 
> Perhaps the downloads on the main site need to be updated?
> 
> > Just download the latest release candidate and you will be golden! https://github.com/ArduPilot/apm_planner/releases
> 
> Where in this link can a DMG file be found? I see no installer on there. I tried downloading the Zip and Tar.gz and neither contained an application or dmg file.
> 
> mboland,
> 
> Where exactly on that page did you find a file that was able to get an installer open on your Mac?
https://ardupilot.org/planner2/docs/installing-apm-planner-2.html
https://firmware.ardupilot.org/Tools/APMPlanner/

drobban, 2020-08-07:
Did a fast and a furious click the links as presented for me according to README and then documentation.
Ended up at https://firmware.ardupilot.org/Tools/APMPlanner/ and that site seems to be outdated?

Is this the official source of builds: https://github.com/ArduPilot/apm_planner/releases

> Where exactly on that page did you find a file that was able to get an installer open on your Mac?
There is a link under the latest release labeled "Assets", click that and you find the latest .DMG

drobban, 2020-08-07:
After reading the comment history, I get the feeling that all reported problems might come from that people are using the old builds served from https://firmware.ardupilot.org/Tools/APMPlanner/.

If so, updated README with current installation information and this issue looks to be resolved?

ferremek, 2021-01-31:
Hey guys,

Have downloaded release 2.0.28-rc1 from https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1. But I cannot install the driver because there is a warning it's not compatible. I have macOS Big Sur 11.1 installed on a MacBook Pro 2018.

Any suggestions how to fix this issue?

Thanks!

billbonney, 2021-01-31:
The driver is now built-in to macOS. You don’t need to install it. Apm  planner 2.0 will work just fine

Iragne, 2023-07-14:
@Arne-W & @billbonney  Do you think this issue can be closed ?

Arne-W, 2023-07-14:
@Iragne Yes you are right - thanks for reviewing.

closed 2023-07-14 by Arne-W