ArduPilot/apm_planner #1173: Home location not updated in GUI on MAV_DO_SET_HOME

AndKe, 2019-06-04:
When home location is updated during flight by GCS plugin (like "movinghome" module for MAVProxy) - AP2 does not update its home location.

Arne-W, 2019-06-09:
@AndKe Should be possible :smiley: 
I had fast look in the Mavlink documentation but could not find the MAV_DO_SET_HOME command?
Do you mean this [https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME)? And do you need full support - setting current and specified location or do you need one of them first?
Just asking cause I am a little bit busy at the moment.... But I will do my best.

AndKe, 2019-06-09:
@Arne-W yes, you linked to correct command, I forgot CMD_ when typing.
ArduPilot does support to set coordinates from parameters 5,6,7 , or current position. It would be neat it AP2 did so too.
That said, the most typical use case it to set custom position (from parameters 5,6,7) - if you could put that together, maybe even I could manage to throw together the remaining support for "set current position" 
Thank you.

Arne-W, 2019-06-10:
@AndKe While I was looking around to find out how to implement this I thought the AP2 should not update the home position when receiving the command but when the UAV acknowledges its execution.
I will try to set up a simulation environment to do some tests first.

Arne-W, 2019-06-10:
@AndKe After debugging a little bit I think [https://mavlink.io/en/messages/common.html#HOME_POSITION](https://mavlink.io/en/messages/common.html#HOME_POSITION) this is the command which should be supported. During simulation this message is sent as soon as the home position is updated. But it seems like there is something fishy going on - it looks like AP2 repeats the MAV_CMD_DO_SET_HOME command.... but I am still checking what is going on...

AndKe, 2019-06-10:
Great to hear that you are looking into this.
I am currently onboard the ship, I may loose internet access after tomorrow. (Just in case I appear to stop responding.)

AndKe, 2019-10-18:
Any progress ?

Arne-W, 2019-11-10:
@AndKe - I accidentally destroyed all the work I had already done :cry: But I will try to fix it again...

Arne-W, 2019-11-17:
@AndKe I am still working on it but I deleted all my work by accident :cry: and so I have to fix this a second time.... and at the moment I don't remember how I fixed it last time....

Arne-W, 2019-11-27:
Should work now :smiley:

closed 2019-11-30 by Arne-W

AndKe, 2019-11-30:
Great!  - thank you.

commit 49643bc3 (2019-11-30): Map Widget: Update Home position icon when vehicle sends HOME_POSITION mavlink message. Fixes #1173