ArduPilot/apm_planner #1178: error trying to install on raspberry PI 3

wx4cb, 2019-08-03:
HI

trying to install the armhf package on a raspberry pi 3 running ubuntu mate. following this thread over on "discuss". I get the error below. Any thoughts?

```
root@GroundStation:~# dpkg -i  apm_planner_2.0.26_xenial_armhf.deb
dpkg: error processing archive apm_planner_2.0.26_xenial_armhf.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 apm_planner_2.0.26_xenial_armhf.deb

```

```
root@GroundStation:~# uname -a
Linux GroundStation 4.15.0-1032-raspi2 #34-Ubuntu SMP PREEMPT Wed Feb 6 11:46:23 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

Arne-W, 2019-08-04:
@wx4cb The armhf package is build for ARM32 systems. You are using an ARM64 (aarch64) System.
Some Arm CPUs support both armhf and aarch64 - but your OS is an aarch64 one so it refuses to install the armhf package.

wx4cb, 2019-08-04:
ok np. that's what i thought.it had something to do with that.

if i install 32 bit support would it work then? or would i have to try and
compile on the pi ?

Arne-W, 2019-08-04:
@wx4cb Yes you could do that.... BUT as the package you are trying to install is REALLY old I would recommend to build the current master on the raspi for aarch64. I will take a while on that system but I think it is worth the effort as the current master has a lot of improvements and bug fixes compared to 2.0.26.

wx4cb, 2019-08-04:
i tried comppiling it 3 times on the pi, but it keeps crashing the pi. i'm
going to setup a cross compiler to do it when i get time

Arne-W, 2019-08-04:
Hmm - perhaps you should try [https://github.com/dockcross/dockcross](https://github.com/dockcross/dockcross). I have heard that it is easy to use - but never tried by myself. :smiley:

wx4cb, 2019-08-15:
still no good. can't cross compile and can't compile on the pi. any other ideas anyone?

dcarpy, 2019-08-16:
@wx4cb Create some swap space on the Pi before compiling. It is likely you are running out of memory.

wx4cb, 2019-08-16:
@dcarpy i'll give that a try thnx.

wx4cb, 2019-08-17:
> @wx4cb Create some swap space on the Pi before compiling. It is likely you are running out of memory.

ok that worked.

Now i have one more question. how do i make a binary package so that we can put it on a resources page so that everyone can get it that's trying to do the same thing.

dcarpy, 2019-08-17:
Use debuild instead of make. You may have to install some additional packages.

dcarpy, 2019-08-17:
Don't forget to close this issue before you are done.  :-)

wx4cb, 2019-08-17:
i tried that, but it appears there's an issue with a libqt5declarative5 package that i can't find in the latest ubuntu mate

wx4cb, 2019-08-17:
```
charlesb@GroundStation:/usr/src/apm_planner$ debuild -us -uc
y
This package has a Debian revision number but there does not seem to be
an appropriate original tar file or .orig directory in the parent directory;
(expected one of apmplanner2_1.0.orig.tar.gz, apmplanner2_1.0.orig.tar.bz2,
apmplanner2_1.0.orig.tar.lzma,  apmplanner2_1.0.orig.tar.xz or apm_planner.orig)
continue anyway? (y/n)  dpkg-buildpackage -rfakeroot -us -uc -ui
dpkg-buildpackage: info: source package apmplanner2
dpkg-buildpackage: info: source version 1.0-cb
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by Charles Blackburn <wx4cb at cfl dot rr dor com>
 dpkg-source --before-build apm_planner
dpkg-buildpackage: info: host architecture arm64
dpkg-checkbuilddeps: error: Unmet build dependencies: git-buildpackage phonon libqt5multimedia5 libqt5positioning5 libqt5webkit5 libqt5webkit5-qmlwebkitplugin | qml-module-qtwebkit libqt5declarative5 qtcreator libsdl1.2-dev flite1-dev libopenscenegraph-dev libqt5webkit5-dev python-serial python-pexpect
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1152:
dpkg-buildpackage -rfakeroot -us -uc -ui failed
charlesb@GroundStation:/usr/src/apm_planner$ sudo apt-get install git-buildpackage phonon libqt5multimedia5 libqt5positioning5 libqt5webkit5 libqt5webkit5-qmlwebkitplugin | qml-module-qtwebkit libqt5declarative5 qtcreator libsdl1.2-dev flite1-dev libopenscenegraph-dev libqt5webkit5-dev python-serial python-pexpect
qml-module-qtwebkit: command not found
E: Unable to locate package libqt5webkit5-qmlwebkitplugin
charlesb@GroundStation:/usr/src/apm_planner$ 
charlesb@GroundStation:/usr/src/apm_planner$ sudo apt-get install git-buildpackage phonon libqt5multimedia5 libqt5positioning5 libqt5webkit5 qml-module-qtwebkit libqt5declarative5 qtcreator libsdl1.2-dev flite1-dev libopenscenegraph-dev libqt5webkit5-dev python-serial python-pexpect
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libqt5declarative5
```

dcarpy, 2019-08-18:
Hmm...when you say latest Ubuntu Mate are you saying 18.04?
https://packages.ubuntu.com/xenial/libqt5declarative5 and
https://packages.ubuntu.com/bionic/allpackages?format=txt.gz

wx4cb, 2019-08-18:
The latest download i beleive is based on bionic.

Ill post a listing wheen i get home later.

Arne-W, 2019-10-17:
Hi all - what is the state of this issue??

wx4cb, 2019-10-17:
> Hi all - what is the state of this issue??

I havent had chance to work on it lately. slammed with a major project at the local airport with work and have had a house move in the middle of it.