ArduPilot/apm_planner #1182: Linux (non-virtual box) Can't update Firmware

naturalmechanics, 2019-09-18:
Issue details

When trying to update firmware by APM 2.0.26, I am getting the error "no valid device found.
check to be sure your APM2.5+or pixhawk/px4 device is plugged in, and drivers are installed."

I had modemmanager de-installed, and i tried with sudo apmplanner2. I used botht he stock version in AUR and compiled one from github directly.
Version

2.0.26 and 2.0.27
Platform

[ x ] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine
Airframe type

N/A
Hardware type

PX4 Cube
Logs

none produced int he console at the time of firmware flash attempt

Arne-W, 2019-10-17:
Hi, @AndKe made a fix to support cubeblack firmware updating. Perhaps you should try it again using the current master. Please let me know if it works - otherwise it would be very cool if you could send me the logfile of the APM-planner which is created in the apmplanner2 folder in you home directory.

Arne-W, 2019-11-10:
@naturalmechanics Hi, have you seen my answer? Does this problem still exist? If not pleas close this issue.

Arne-W, 2020-02-27:
closing this as its a known problem and we are still working on a good solution. See #1171 and #1111

closed 2020-02-27 by Arne-W