ArduPilot/apm_planner #1186: Remove/Hide APMToolBar

benb0jangles, 2019-11-05:
Simple question
What is the quickest most simple way to remove/hide the APMToolBar widget from MainWindow.ui

Thanks

Arne-W, 2019-11-10:
You could build the Planner with #define QGC_TOOLBAR_ENABLED. It will remove the toolbar object. after that you will have to fix some linker errors because several pages refer to the toolbar.
I think this will be the fastest way to remove the toolbar.
BUT why do you want to remove it???

benb0jangles, 2019-11-11:
Hi Arne, Yea I thought as much- that if I comment out the toolbar I will have to go through the other script pages and correct errors too. 
Removing the APMtoolbar (the one with the icons not the menu bar) means there will be more more flight data viewable on my 640x360 fpv screen. 
The screen I edited so far, it's not too bad, It's really just about making the best of available screen space.

It is clear that to make efficient use of the screen space removing the APMToolbar might be done. It is ok to do this because the standard menu bar > File > Edit > View  this also does the same thing as APMToolbar so no need to have twice

I am building for Raspberry Pi + Cheap 7" LCD TV Screen.
benefits over Android Tablet/Laptop:
- Cheap ($10 screen + Pi)
- The Screen has two channel switch, ch1 for APMPlanner, ch2 for FPV view. 
- Small linux computer means I am able to have more practical use
- It is smaller/portable than laptop (screen mounts onto Flysky radio transmitter)
- USB + GPIO I am able to use many tx/rx modules (Android is limited)
- I am using c2 fpv with Movidius Neural Compute stick live video object identification 
- Also OpenCV Object Tracking

Arne-W, 2021-05-18:
No changes for a long time - closing this issue.

closed 2021-05-18 by Arne-W