ArduPilot/apm_planner #1190: cannot download apm planner on mac os 10.15.1

Johanth3108, 2019-12-03:
when i download the zip file and try to install the software i cannot install it, my os is catalina.

yuruyenadam, 2020-01-19:
i have same issue what should i do?

Arne-W, 2020-02-01:
Which zip file are you talking about? 
Use this: [https://firmware.ardupilot.org/Tools/APMPlanner/daily/2019-03-08/apm_planner_2.0.27-rc1-7-g299ff23_osx.dmg](https://firmware.ardupilot.org/Tools/APMPlanner/daily/2019-03-08/apm_planner_2.0.27-rc1-7-g299ff23_osx.dmg)

Arne-W, 2020-02-21:
Is this still an issue? If not could you please close it?!

closed 2020-02-27 by Arne-W

scientistbala, 2021-01-08:
hi sir,
I have the same issues, my OS is Catalina ver 10.15.7.
I tried to install mission planner its showing error.
I attached the image below, how to fix this
[image]