ArduPilot/apm_planner #1193: When will there be the next stable release?

wucke13, 2020-01-03:
`2.0.27-rc1` is over a year old and only a release candidate
`2.0.26` is close two 1 1/2 years old now

billbonney, 2020-01-04:
Essentially 2.0.27-rc1 is the latest stable release. Another compiled release would be nice. I'll leave that for others to comment on.

wucke13, 2020-01-04:
I actually don't care about a binary release. I maintain the the NixOS apmplanner 2 package, and would like to update it. However I don't feel lucky too just choose "random" state of master, therefore I would like upstream (this package) to release at least every 9 months or so (preferable more often).

billbonney, 2020-01-04:
Ok. Need to ask @Arne-W what he considers a stable version.

Arne-W, 2020-01-04:
Hi all, current master **is** stable! I thought we should make a release too and did some testing to create an appimage package to be a little more distribution independent but have not finished yet.
I did not test if everything works with the new released arducopter 4.0 version - that is still on my list.

anbello, 2020-01-04:
I think that for a new stable version it should be considered the change in parameters from Copter 4.0.
Something to implement here:
https://github.com/ArduPilot/apm_planner/blob/master/src/ui/configuration/CopterPidConfig.cc#L171

liike this:

```
    if(version.majorNumber() == 3)
    {
        if(version.minorNumber() >= 6)
        {
            QLOG_DEBUG() << "Using parameter set for ArduCopter 3.6+";
            setupPID_APM_36();
        }
        else if(version.minorNumber() >= 4)
        {
            QLOG_DEBUG() << "Using parameter set for ArduCopter 3.4+";
            setupPID_APM_34();
        }
    }
    else if(version.majorNumber() == 4)
    {
	if(version.minorNumber() >= 0)
        {
            QLOG_DEBUG() << "Using parameter set for ArduCopter 4.0+";
            setupPID_APM_40();
        }
    }
    else
    {
        QLOG_DEBUG() << "Using default parameter set";
        setupPID_Default();
    }
```

and then a new function `setupPID_APM_40()` witch will be mainly similar to `setupPID_APM_36()` but with some difference, for now I see difference in ATC_RAT_\*_FILT that split in three params ATC_RAT_\*_FLTD, ATC_RAT_\*_FLTE, ATC_RAT_\*_FLTT and the same for PSC_ACCZ_FILT.

anbello, 2020-01-04:
Sorry @Arne-W i didn't see your last comment when I write mine.

wucke13, 2020-01-05:
Well that all sounds promising :smile:

Arne-W, 2020-01-15:
takes longer than expected - but I am working on it....

Arne-W, 2020-01-24:
okay - this is done now... but there is still #1171 which needs to be fixed before we can make a new release.....

commit b7cebba7 (2020-01-24): Tuning Setup: Added support for Arducopter 4.0 Fixes #1193

* Added new filters to extended tuning
* Added feed forwared paramter to extended tuning
* Updated tune & rc options dropdown lists
* Updated Parameters in basic and extended tuning
closed 2020-01-24 by Arne-W