ArduPilot/apm_planner #1196: Adding new flight mode

Bggopal, 2020-02-01:
Hi guys,
I am trying to add a new flight mode in Copter-3.5 using the documentation provided in ("https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode-35.html"), but even after updating every field (by creating a new flight mode similar to stabilize) i couldn't see it in APM planner, i using APM planner v2.0.23 in linux. the most searchers where relating to a file in mission planner "ParameterMetaData.xml" but for me here it is "apm.pdef.xml" . Even though i edited it, it does not work. Was i on the right way, or where did i do mistake, if you any one have good experience, please provide your help here. Thank you.

Git repo for the changes i did "https://github.com/Bggopal/ardupilot/tree/Copter-3.5"

[x] Pixhawk
[x] Copter
[image]

It would be great of you to help Mr. @muramura

muramura, 2020-02-02:
@Bggopal san.
I update two files to add a new flight mode.
I can find two files on ZIGZAG.
I add a new flight mode to these two files.

[image]

Bggopal, 2020-02-04:
Thank you Mr. @muramura  ,  it worked, but one more small issue is that after trying to implement the created new mode it shows FAILURE: failed cmd 11, in apm planner. I Think it is some part error with firmware programming. What i need to change that. Though i have changed the files for new flight mode in  control_newflightmode.cpp and called the functions crtly.
[image]

muramura, 2020-02-11:
@barthess san.
I just added the definition and processing to APM Planner2 only, the flight controller does not recognize that number.
Add definitions and processing to the flight controller as well.
I defined TESTMODE (127), specified POSHOLD for processing, and built the firmware.
I specified TESTMODE from APM PLANNER2 and confirmed that the flight mode became TESTMODE.
I was able to fly TESTMODE at SITL.

[image]

[image]
[image]

Arne-W, 2020-02-21:
So - cool you made it but is this still an issue? If not would you please close this issue? If you made a change which should be merged create a PR.

closed 2020-02-22 by Bggopal