ArduPilot/apm_planner #1197: map is not loading in apm planner2.0

vishal-raj15, 2020-02-21:
map is not loading and map files  in flight planner not loading

Arne-W, 2020-02-21:
Which version? Did you build from master? Which plattform? Linux? Macos?

vishal-raj15, 2020-02-22:
APM planner 2.0 in ubuntu linux

Arne-W, 2020-02-24:
@vishal-raj15 Try this [DOWNLOAD](https://www.dropbox.com/s/ywtbtx6zrcweiwo/APM-Planner_2-2.0.27-rc1-x86_64.AppImage?dl=0)
It is an AppImage of a current master build. On ubuntu an AppImage should work with a simple double click.
Known issue of this build: Flashing new firmwares does not work.

vishal-raj15, 2020-02-24:
@Arne-W  thanks , but after running that .appimage it shows:

[image]

Arne-W, 2020-02-25:
@vishal-raj15 Hmm this is strange.... However could you please change the map type using the options button on the left below the map window. Moreover you could check if there is an apmplanner2 folder in you home directory. I should contain some logfiles and a folder named mapscache which should contain a file called Data.qmdb. All this locations have to be writable. Take a look into the logfiles an check for anything strange.
You can also try deleting the APM Planner.ini file you will find in your home .config/ardupilot/

vishal-raj15, 2020-02-25:
@Arne-W  i have changed the logfiles and mapchache to writable but yet its the same condition

Arne-W, 2020-02-26:
Did you try to change the used map type?? Did you try to delete the APM Planner.ini file??

vishal-raj15, 2020-02-26:
@Arne-W  yes but still

Arne-W, 2020-02-27:
Ah all right - you have a problem with your network connection. See the error line in your log it says timeout when trying to fetch pdef.xml file. Moreover this is not the version I created as this version is 2.0.26.
I deleted the old appimage from my dropbox and created a new one for you. Here is the Link:
[Download AppImage](https://www.dropbox.com/s/1f21s7jg06t29i2/APM-Planner_2-9d8c1c370-x86_64.AppImage?dl=0)

vishal-raj15, 2020-02-27:
@Arne-W  i download this .appimage and then make it executable and after clicking on apm planner icon only black screnn appears(just showing loading tiles) , do i need to install some other files ?

vishal-raj15, 2020-02-27:
[image]

Arne-W, 2020-02-29:
@vishal-raj15 It looks like you did not start the application from the AppImage. It should be possible to start that image without any installation - everything needed is contained in that package. Simply do a double click on the AppImage after you made it executable. If this does not work extract the image by using the --appimage-extract commandline option. It will create a folder named sqashfs-root. Change into that folder and then into usr/bin. There you will find an apmplanner2 binary you can start.
To check if you started the right version go to menubar -> Help -> Show debug console. At the bottom of the Debug Output window you will find a field called Commit it should contain this string: 2.0.27-rc1-33-g9d8c1c3.

vishal-raj15, 2020-03-01:
@Arne-W  in that commit version is : 2.0.26-1-g05c4d11 , but i downloaded  2.0.27-rc1-33-g9d8c1c3. version appimage link given by you

Arne-W, 2020-03-01:
@vishal-raj15 Sorry but you are still executing the wrong version. Your are using the installed version **NOT**  the appImage version. Perhaps you should remove the installed version form your machine.
Try to use the AppImage file as I described above - just make the file executable and start it by double clicking or command line.

vishal-raj15, 2020-03-01:
@Arne-W  now i changed that version to 2.0.27-rc1-33-g9d8c1c3 and made it and mapchache executable but only black image in place of map

Arne-W, 2020-03-01:
@vishal-raj15 could you please post the logfile

vishal-raj15, 2020-03-02:
@Arne-W from where can i get that logfiles there is no option of logfiles in the apm planner's tabs

Arne-W, 2020-03-07:
@vishal-raj15 in your home directoy the should be an apmplanner2 directory. Inside this directory you will find files called apmlog_xxxxxxx.txt this are the logfiles.

vishal-raj15, 2020-03-08:
[apmlog_20200302.txt](https://github.com/ArduPilot/apm_planner/files/4303105/apmlog_20200302.txt)

vishal-raj15, 2020-03-08:
@Arne-W  i found two .txt files 20200302 and 20200301.txt

vishal-raj15, 2020-03-08:
**************************************************

[20200302 22:22:11.720 INFO ] - QGCCore::initialize()
[20200302 22:22:11.720 INFO ] - Current Build Info
[20200302 22:22:11.720 INFO ] - Git Hash: 9d8c1c37016f79d565a66d82fe3b059e0ecaaf4e
[20200302 22:22:11.720 INFO ] - Git Commit: 2.0.27-rc1-33-g9d8c1c3
[20200302 22:22:11.720 INFO ] - APPLICATION_NAME: "APM Planner"
[20200302 22:22:11.720 INFO ] - APPLICATION_VERSION: "v2.0.27-rc1"
[20200302 22:22:11.720 INFO ] - APP_PLATFORM: ubuntu64
[20200302 22:22:11.720 INFO ] - APP_TYPE: stable
[20200302 22:22:11.976 INFO ] - Start Link Manager
[20200302 22:22:11.976 DEBUG] - Create MAVLinkDecoder:  MAVLinkDecoder(0x3136fa0)
[20200302 22:22:11.980 INFO ] - Start UAS Manager
[20200302 22:22:12.021 DEBUG] - Creating MainWindow
[20200302 22:22:12.250 INFO ] - Creating SubMainWindow:  SubMainWindow(0x31943e0)
[20200302 22:22:12.251 INFO ] - setting objectName:  "VIEW_MISSION"
[20200302 22:22:12.330 INFO ] - Creating SubMainWindow:  SubMainWindow(0x32c6e20)
[20200302 22:22:12.330 INFO ] - setting objectName:  "VIEW_FLIGHT"
[20200302 22:22:12.364 INFO ] - Creating SubMainWindow:  SubMainWindow(0x333b600)
[20200302 22:22:12.364 INFO ] - setting objectName:  "VIEW_HARDWARE_CONFIG"
[20200302 22:22:12.633 DEBUG] - fillPortsInfo 
[20200302 22:22:12.935 INFO ] - Creating SubMainWindow:  SubMainWindow(0x33b7460)
[20200302 22:22:12.935 INFO ] - setting objectName:  "VIEW_SOFTWARE_CONFIG"
[20200302 22:22:12.952 DEBUG] - Disable Controls
[20200302 22:22:12.981 INFO ] - BasicPidConfig Class only supports ArduPilotMegaMAV vehicles.
[20200302 22:22:13.166 INFO ] - Creating SubMainWindow:  SubMainWindow(0x4488b70)
[20200302 22:22:13.166 INFO ] - setting objectName:  "VIEW_ENGINEER"
[20200302 22:22:13.176 INFO ] - Creating SubMainWindow:  SubMainWindow(0x44eaaa0)
[20200302 22:22:13.176 INFO ] - setting objectName:  "VIEW_SIMULATOR"
[20200302 22:22:13.224 DEBUG] - MODE: 0
[20200302 22:22:13.230 DEBUG] - "D|PREFLIGHT"
[20200302 22:22:13.230 DEBUG] - MODE: 80
[20200302 22:22:13.230 DEBUG] - "D|STABILIZED"
[20200302 22:22:13.230 DEBUG] - MODE: 64
[20200302 22:22:13.230 DEBUG] - "D|MANUAL"
[20200302 22:22:13.231 DEBUG] - MODE: 88
[20200302 22:22:13.231 DEBUG] - "D|VECTOR|STABILIZED"
[20200302 22:22:13.231 DEBUG] - MODE: 92
[20200302 22:22:13.231 DEBUG] - "D|AUTO|STABILIZED"
[20200302 22:22:13.232 DEBUG] - MODE: 66
[20200302 22:22:13.232 DEBUG] - "D|TEST"
[20200302 22:22:13.302 DEBUG] - QUrl("file:///home/vishal/.local/share/Trash/files/squashfs-root/usr/bin/qml/PrimaryFlightDisplayQML.qml")
[20200302 22:22:13.620 DEBUG] - QML Status: QQuickView::Ready
[20200302 22:22:13.623 DEBUG] - QUrl("file:///home/vishal/.local/share/Trash/files/squashfs-root/usr/bin/qml/PrimaryFlightDisplayQML.qml")
[20200302 22:22:13.644 DEBUG] - QML Status: QQuickView::Ready
[20200302 22:22:13.653 INFO ] - UASActionsWidget creating  UASActionsWidget(0x492f530)
[20200302 22:22:13.656 DEBUG] - QuickView: load settings
[20200302 22:22:13.657 DEBUG] - QuickView: loading key:  "GCS GPS.GPS Fix ()"  type: "text"
[20200302 22:22:13.667 DEBUG] - QuickView: loading key:  "GCS GPS.GPS HDOP ()"  type: "text"
[20200302 22:22:13.668 DEBUG] - QuickView: loading key:  "GCS GPS.GPS Sats (n)"  type: "text"
[20200302 22:22:13.670 DEBUG] - QuickView: loading key:  "GCS Metric.Alt MSL (m)"  type: "text"
[20200302 22:22:13.672 DEBUG] - QuickView: loading key:  "GCS Metric.Alt REL (m)"  type: "text"
[20200302 22:22:13.674 DEBUG] - QuickView: loading key:  "GCS Status.Battery (%)"  type: "text"
[20200302 22:22:13.676 DEBUG] - QuickView: loading key:  "GCS Status.Climb (m/s)"  type: "text"
[20200302 22:22:13.678 DEBUG] - QuickView: loading key:  "GCS Status.Current (A)"  type: "text"
[20200302 22:22:13.680 DEBUG] - QuickView: loading key:  "GCS Status.Pitch (deg)"  type: "text"
[20200302 22:22:13.684 DEBUG] - QuickView: loading key:  "GCS Status.Roll (deg)"  type: "text"
[20200302 22:22:13.687 DEBUG] - QuickView: loading key:  "GCS Status.Voltage (V)"  type: "text"
[20200302 22:22:13.690 DEBUG] - QuickView: loading key:  "GCS Status.Yaw (deg)"  type: "text"
[20200302 22:22:13.696 DEBUG] - QuickView: load settings END
[20200302 22:22:13.751 INFO ] - Using Alsa Audio driver
[20200302 22:22:13.762 WARN ] - Dir directory tmp_audio exists
[20200302 22:22:13.766 DEBUG] - qmlBaseDir "/home/vishal/.local/share/Trash/files/squashfs-root/usr/bin"
[20200302 22:22:13.766 DEBUG] - QUrl("file:///home/vishal/.local/share/Trash/files/squashfs-root/usr/bin/qml/ApmToolBar.qml")
[20200302 22:22:13.920 DEBUG] - APMToolBar: button label SELECTED
[20200302 22:22:13.932 DEBUG] - QML Status: QQuickView::Ready
[20200302 22:22:13.933 DEBUG] - Setting for  "USER_DONATED" : QVariant(QString, "false")
[20200302 22:22:13.934 DEBUG] - Set Advanced Mode false USER_DONATED:false
[20200302 22:22:13.934 DEBUG] - Donate visible
[20200302 22:22:13.935 INFO ] - NETWORK_PROXY: Disabling System Network Proxies
[... 43 lines cut ...]
[20200302 22:23:19.813 DEBUG] - Downloading: 0 / 0
[20200302 22:23:19.814 DEBUG] - AutoUpdateCheck::httpFinished()
[20200302 22:28:34.935 DEBUG] - APMToolBar: GRAPHS SELECTED
[20200302 22:28:34.936 DEBUG] - APMToolBar: clear selected buttons
[20200302 22:28:34.937 DEBUG] - APMToolBar: FLIGHT DATA unselected
[20200302 22:28:34.937 DEBUG] - APMToolBar: GRAPHS unselected
[20200302 22:28:35.067 DEBUG] - APMToolBar: GRAPHS SELECTED
[20200304 10:29:43.576 DEBUG] - Delete Widget  QMenuBar(0x31a6780, name="menuBar")
[... cut ...]
[20200304 10:29:44.158 DEBUG] - Delete DockWidget  QDockWidget(0x4c54360)
[20200304 10:29:44.159 DEBUG] - Destory APM Toolbar
[20200304 10:29:44.167 DEBUG] - Delete Widget  CommConfigurationWindow(0x53021e0, name="commSettings")
[20200304 10:29:44.171 DEBUG] - Delete Widget  CommConfigurationWindow(0x5342730, name="commSettings")
[20200304 10:29:44.177 DEBUG] - Destroy MAVLinkDecoder:  MAVLinkDecoder(0x3136fa0)
[20200304 10:29:44.177 DEBUG] - save tlog dir to: "/home/vishal/apmplanner2/tlogs"
[20200304 10:29:44.275 INFO ] - ~GAudioOutput()

Arne-W, 2020-03-08:
@vishal-raj15 Did you check you network connectivity? It looks like the ApmPlanner has no internet connection. Perhaps you have a problem with the permissions? The problem seems to be somehow related to you system setup.
Without internet connection you won't have any map.

vishal-raj15, 2020-03-08:
@Arne-W  i can't find how to connect , when ever i click to connect icon in 
apm planner software on top right it says error opening port no such file

Arne-W, 2020-03-08:
The connect icon is used to connect to an UAV.
Does your computer have a working intenet connection?
Please enter `ping google.com` in a shell window and post the result.

vishal-raj15, 2020-03-09:
@Arne-W , yes!

 vishal@vishal-pc  ~  ping google.com
PING google.com (172.217.166.14) 56(84) bytes of data.
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=1 ttl=54 time=22.8 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=2 ttl=54 time=25.8 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=3 ttl=54 time=25.3 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=4 ttl=54 time=27.6 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=5 ttl=54 time=26.7 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=6 ttl=54 time=27.3 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=7 ttl=54 time=27.9 ms
64 bytes from del03s17-in-f14.1e100.net (172.217.166.14): icmp_seq=8 ttl=54 time=32.5 ms

Arne-W, 2020-03-11:
@vishal-raj15 okay I have no Idea whats going on - nobody (I know) has this problem. Last test from my side please send me the output of `ifconfig` and of `route`

vishal-raj15, 2020-03-12:
@Arne-W  

 vishal@vishal-pc  ~  ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 98:28:a6:0e:d1:4f brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 70:c9:4e:8f:6e:61 brd ff:ff:ff:ff:ff:ff
    inet 10.8.9.38/16 brd 10.8.255.255 scope global dynamic noprefixroute wlp3s0
       valid_lft 17172sec preferred_lft 17172sec
    inet6 fe80::7842:e686:74cf:8695/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

vishal-raj15, 2020-03-12:
@Arne-W 

vishal@vishal-pc  ~  ip route
default via 10.8.0.1 dev wlp3s0 proto dhcp metric 20600 
10.8.0.0/16 dev wlp3s0 proto kernel scope link src 10.8.9.38 metric 600 
169.254.0.0/16 dev wlp3s0 scope link metric 1000

vishal-raj15, 2020-03-29:
@Arne-W  apm planner works on linux without proxy , but not with proxy

Arne-W, 2020-03-30:
@vishal-raj15 Nice to hear you found it. I never tried using the planner with a proxy. I will try to figure out what goes wrong in that case.
Thank you for informing us!

kvlahromei, 2021-06-07:
Same for Windows platform. it ignores system proxy and struggles to fetch map tiles.

commit 1f693296 (2025-07-02): Fixes #1197: Update map tile download to https

- Updates maps urls to use https which fixes issue on linux/windows
- NOTE: Does not fix non-functioning map tile providers.
closed 2025-07-02 by billbonney