ArduPilot/apm_planner #1199: APM Planner 2.0 on macOS Catalina (10.15.3)

ferranroigtio, 2020-03-19:
Just wanted to share that version 26 doesn't work on macOS Catalina. However, version 25 works fine and does not require the FTDI Driver install. Never tried beta version 27.

mboland, 2020-03-19:
Just to confirm
trying to open 2.0.26 on MacOS Catalina causes a crash

Process:               apmplanner2 [2832]
Path:                  /Applications/AutoPilots/APM Planner 2.0.26.app/Contents/MacOS/apmplanner2
Identifier:            org.ardupilot.apmplanner2
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           apmplanner2 [2832]
User ID:               501

Date/Time:             2020-03-20 07:52:57.339 +1000
OS Version:            Mac OS X 10.15.4 (19E264b)
Report Version:        12
Anonymous UUID:        D11B06CF-C0DE-5370-97B2-CE246A81EC7F

Time Awake Since Boot: 1400 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries

mboland, 2020-03-19:
Version 2.0.27-rc1 does run OK on MacOS Catalina

ferranroigtio, 2020-03-20:
Cheers, Mike. Would you recommend me to upgrade to version 2.0.27? Or stick to 2.0.25?

mboland, 2020-03-20:
Considering the massive changes in Catalina I would recommend you upgrade APM2 to the 2.0.27

ferranroigtio, 2020-03-20:
That's cool, thanks!

closed 2020-03-20 by ferranroigtio