ArduPilot/apm_planner #1204: set home=0 on start apmplanner2

Andr3g2, 2020-04-27:
Hallo,

After each start of the app the coordinate WP0=0,0 (HOME=0,0)
LAT en LON is zero which indicates a location somewhere in the sea before Africa.
only later, when the enter menu:flightplan:editwp <refresh>, the value is correct.
details see below.

Whats going wrong?
This problem has been around for a long time.  (2.0.24, 2.0.27)

I think a WP should never be NULL.
If write WP to device there should be print a warning if WP=0,0

regard Andre

details:

apmplanner v2.0.27-rc1

- s1
$ cat mission.rov.test2.txt
QGC WPL 110
0       1       0       16      0       0       0       0       51.5100000000000001     14.1400000000000001     111.111000000000000     1
1       0       3       16      0       0       0       0       51.5200000000000001     14.1500000000000001     0       1
2       0       3       16      0       0       0       0       51.5300000000000001     14.1600000000000001     0       1
3       0       3       16      0       0       0       0       51.5100000000000001     14.1700000000000001     0       1
4       0       3       16      0       0       0       0       51.5100000000000001     14.1500000000000001     0       1

- s2
$ apmplanner2.sh | egrep -i "wp|way|point|status|xml"
[20200427 19:33:35.490 INFO ] - APPLICATION_VERSION: "v2.0.27-rc1"
[20200427 19:33:36.868 DEBUG] - Writing ( "https://autotest.ardupilot.org/Parameters/ArduPlane/apm.pdef.xml" ) to ( "/home/admin1/apmplanner2/ardupilotmega/arduplane.pdef.xml" )
[20200427 19:33:52.033 DEBUG] - Using ( "/home/admin1/apmplanner2/ardupilotmega/ardurover.pdef.xml" ) for parameters
[20200427 19:33:52.090 INFO ] - updateWaypointList:
[20200427 19:33:52.093 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.095 DEBUG] - UAS: new system_status 2
[20200427 19:33:52.129 INFO ] - updateWaypointList:
[20200427 19:33:52.149 DEBUG] - handleWaypointCount() - Number of waypoints to fetch is  5
[20200427 19:33:52.150 INFO ] - STATUS TEXT: 1 : "ArduRover v2.52-beta (5a6394ed)"
[20200427 19:33:52.303 INFO ] - updateWaypointList:
[20200427 19:33:52.303 INFO ] -   WP: 0
[20200427 19:33:52.303 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.304 DEBUG] - handleWaypoint() - Received waypoint  0
[20200427 19:33:52.349 INFO ] - STATUS TEXT: 2 : "Warming up ADC..."
[20200427 19:33:52.406 INFO ] - updateWaypointList:
[20200427 19:33:52.407 INFO ] -   WP: 0
[20200427 19:33:52.407 INFO ] -   WP: 1
[20200427 19:33:52.407 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.408 DEBUG] - handleWaypoint() - Received waypoint  1
[20200427 19:33:52.457 INFO ] - STATUS TEXT: 1 : "<startup_ground> GROUND START"
[20200427 19:33:52.519 INFO ] - updateWaypointList:
[20200427 19:33:52.519 INFO ] -   WP: 0
[20200427 19:33:52.520 INFO ] -   WP: 1
[20200427 19:33:52.520 INFO ] -   WP: 2
....
[20200427 19:33:52.842 INFO ] - STATUS TEXT: 2 : "Beginning INS calibration; do not move vehicle"
[20200427 19:33:53.737 DEBUG] - Param: "WP_RADIUS" :  QVariant(double, 2)
[20200427 19:33:54.714 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20200427 19:33:56.136 INFO ] - STATUS TEXT: 1 : "\n\n Ready to drive."
[20200427 19:33:56.152 INFO ] - "New current waypoint 0"

- s3
 menu:flightplan:onboardwp
  0 HOME (lat. 0.000000, lon 0.000000, alt 0.00) yaw: 0;

- s4
 menu:flightplan:onboardwp <refresh>
  0 HOME (lat. 51.1500000, lon 14.1400000, alt 111.11) yaw: 0;

- s5
 menu:flightplan: <go home>
  -> fly to africa

- s6
  menu:flightplan:editwp
   0 HOME (lat. 0.000000, lon 0.000000, alt 0.00) yaw: 0;

- s7
  menu:flightplan:editwp <refresh>
  0 HOME (lat. 51.1500000, lon 14.1400000, alt 111.11) yaw: 0;

- s8
 menu:flightplan: <go home>
  -> fly to MYHOME (WP 0)

Andr3g2, 2020-05-08:
sorry i have see any lines are unreadable.
edit: replace htmltag with "keyword"

Hallo,

After start apmplanner2 the coordinate are WP0=0,0 (HOME=0,0)
LAT en LON is zero which indicates a location somewhere in the sea before Africa
.
later, if the enter menu:flightplan:editwp:"refresh", the value is correct.
details see below.

Whats going wrong?
This problem has been around for a long time.  (2.0.24, 2.0.27)

my mind:
I think a WP should be never NULL.
If write WP to device, there should be print a warning if WP=0,0

regards Andre

details:

apmplanner v2.0.27-rc1

- s1
$ cat mission.rov.test2.txt 
QGC WPL 110
0       1       0       16      0       0       0       0       51.5100000000000
001     14.1400000000000001     111.111000000000000     1
1       0       3       16      0       0       0       0       51.5200000000000
001     14.1500000000000001     0       1
2       0       3       16      0       0       0       0       51.5300000000000
001     14.1600000000000001     0       1
3       0       3       16      0       0       0       0       51.5100000000000
001     14.1700000000000001     0       1
4       0       3       16      0       0       0       0       51.5100000000000
001     14.1500000000000001     0       1

 
- s2
$ apmplanner2.sh | egrep -i "wp|way|point|status|xml"
[20200427 19:33:35.490 INFO ] - APPLICATION_VERSION: "v2.0.27-rc1"
[20200427 19:33:36.868 DEBUG] - Writing ( "https://autotest.ardupilot.org/Parame
ters/ArduPlane/apm.pdef.xml" ) to ( "/home/admin1/apmplanner2/ardupilotmega/ardu
plane.pdef.xml" )
[20200427 19:33:52.033 DEBUG] - Using ( "/home/admin1/apmplanner2/ardupilotmega/
ardurover.pdef.xml" ) for parameters
[20200427 19:33:52.090 INFO ] - updateWaypointList: 
[20200427 19:33:52.093 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.095 DEBUG] - UAS: new system_status 2
[20200427 19:33:52.129 INFO ] - updateWaypointList: 
[20200427 19:33:52.149 DEBUG] - handleWaypointCount() - Number of waypoints to f
etch is  5
[20200427 19:33:52.150 INFO ] - STATUS TEXT: 1 : "ArduRover v2.52-beta (5a6394ed
)"
[20200427 19:33:52.303 INFO ] - updateWaypointList: 
[20200427 19:33:52.303 INFO ] -   WP: 0
[20200427 19:33:52.303 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.304 DEBUG] - handleWaypoint() - Received waypoint  0
[20200427 19:33:52.349 INFO ] - STATUS TEXT: 2 : "Warming up ADC..."
[20200427 19:33:52.406 INFO ] - updateWaypointList: 
[20200427 19:33:52.407 INFO ] -   WP: 0
[20200427 19:33:52.407 INFO ] -   WP: 1
...
[20200427 19:33:52.798 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200427 19:33:52.799 DEBUG] - handleWaypoint() - Received waypoint  4
[20200427 19:33:52.819 DEBUG] - handleWaypoint() - Received all waypoints 
[20200427 19:33:52.842 INFO ] - STATUS TEXT: 2 : "Beginning INS calibration; do 
not move vehicle"
[20200427 19:33:53.737 DEBUG] - Param: "WP_RADIUS" :  QVariant(double, 2)
[20200427 19:33:54.714 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20200427 19:33:56.136 INFO ] - STATUS TEXT: 1 : "\n\n Ready to drive."
[20200427 19:33:56.152 INFO ] - "New current waypoint 0"

- s3
  menu:flightplan:onboardwp
  0 HOME (lat. 0.000000, lon 0.000000, alt 0.00) yaw: 0;

- s4
  menu:flightplan:onboardwp:"refresh"
  0 HOME (lat. 51.1500000, lon 14.1400000, alt 111.11) yaw: 0;

- s5
  menu:flightplan:"go home"
  -> fly to africa

- s6
  menu:flightplan:editwp
   0 HOME (lat. 0.000000, lon 0.000000, alt 0.00) yaw: 0;

- s7
  menu:flightplan:editwp:"refresh"
  0 HOME (lat. 51.1500000, lon 14.1400000, alt 111.11) yaw: 0;

- s8
  menu:flightplan:"go home"
  -> fly to MYHOME (real WP 0)

Arne-W, 2020-05-10:
The question is always which position should be used by default? As soon as we have seen an vehicle sending a valid home position its used and stored. But as long as we have not seen any valid position which position shall be used for home?

Arne-W, 2020-05-17:
@Andr3g2 Have you seen my comment??

Andr3g2, 2020-05-31:
Dear Arne,

Thanks for your answer.
I dont understand workflow.
on issue  #1189 written: receiving MAVLINK_MSG_ID_HOME_POSITION the UAV home position
but i have not found this.
I have 2 different controller. seem different behavior.
(ardupilot, minipix)

sometimes open a windows with following message: 
Set a new home position for vehicle MAV 001
Do you want to set a new origin? Waypoints defined in the local frame will be shifted in their physical location
seem correspond with following log:
[20200531 22:26:10.078 INFO ] - STATUS TEXT: 6 : "EKF2 IMU0 Origin set to GPS"
[20200531 22:26:55.935 INFO ] - STATUS TEXT: 6 : "EKF2 IMU0 is using GPS"

Scenario1
- start applanner
- start minipix (connect on usb)
- after 5 sec <connect>
  - symbol Home (black)  visible (correct position)
  - wait on sat...
  - WP: 
     0 Home (lat 0.000, lon 0.0...
     ....
     5  Return to launch location
  -> i cannot find a way to fix it.
[20200531 21:53:17.241 DEBUG] - UPDATE WP LIST IN 2D MAP CALLED FOR UAS 1
[20200531 21:53:17.248 DEBUG] - handleWaypoint() - Received waypoint  5
[20200531 21:53:17.268 DEBUG] - handleWaypoint() - Received all waypoints
[20200531 21:53:17.320 DEBUG] - Param: "WPNAV_SPEED" :  QVariant(double, 500)
[20200531 21:53:17.321 DEBUG] - UASAction setting speed spin box from  "WPNAV_SPEED"
[20200531 21:53:17.322 DEBUG] - Param: "WPNAV_RADIUS" :  QVariant(double, 200)
[20200531 21:53:17.326 DEBUG] - Param: "WPNAV_SPEED_UP" :  QVariant(double, 250)
[20200531 21:53:17.329 DEBUG] - Param: "WPNAV_SPEED_DN" :  QVariant(double, 110)
[20200531 21:53:17.341 DEBUG] - Param: "WPNAV_RADIUS" :  QVariant(double, 200)
[20200531 21:53:17.342 DEBUG] - Param: "WPNAV_RADIUS" :  QVariant(double, 200)
[20200531 21:53:17.350 DEBUG] - Param: "WPNAV_ACCEL" :  QVariant(double, 100)
[20200531 21:53:17.351 DEBUG] - Param: "WPNAV_ACCEL_Z" :  QVariant(double, 100)
[20200531 21:53:17.352 DEBUG] - Param: "WPNAV_RFND_USE" :  QVariant(int, 1)
[20200531 21:53:18.352 DEBUG] - Param: "RALLY_INCL_HOME" :  QVariant(int, 1)
[20200531 21:53:18.510 INFO ] - "New current waypoint 0"
[20200531 21:53:19.064 DEBUG] - Param: "WP_NAVALT_MIN" :  QVariant(double, 0)
[20200531 21:53:23.061 DEBUG] - Param: "SRTL_POINTS" :  QVariant(int, 150)
[20200531 21:53:26.306 INFO ] - STATUS TEXT: 2 : "PreArm: Throttle below Failsafe"

Scenario2
- start applanner
- start minipix (connect on usb)
- after 5 sec <connect>
  - symbol Home (black)  visible (correct position)
  - wait on sat...
  - WP: 
     0 Home (lat 0.000, lon 0.0...
     ....
     5  Return to launch location
 
- stop applanner
- start applanner
- after 5 sec <connect>
  - symbol Home (black)  visible (correct position)
  - wait on sat...
  - WP:
     0 Home (lat 0.000, lon 0.0...
     ....
     5  Return to launch location

  - windows: Set a new home position for vehicle MAV 001 -> yes
  - WP:
     0 Home (lat 51... real value)
     ...
     5 RTL (0.0 , 0.0 )   -> fly to africa

regards Andre