ArduPilot/apm_planner #1209: uas: In AC4.0.0 or later, automatic flight plan display, update, writing, and reading are not completed.

muramura, 2020-05-16:
I used a master containing #1207 to check the automatic flight plan reading from the AC4.0.0 release version.
At startup, T YPE is set to an indefinite value for list acquisition and WP acquisition, so it is not displayed.

[image]

[image]

I was able to show it at startup by making the same fix as #1207.

[image]

===
However, when the WP becomes too much, a timeout occurs and the game is finished in the middle.
So, by increasing the timeout time and the number of retries, we received all the WPs.

What should I do about this time and frequency value?
#define PROTOCOL_TIMEOUT_MS 180000 //2000 ///< maximum time to wait for pending messages until timeout
#define PROTOCOL_MAX_RETRIES 90 // 5 ///< maximum number of send retries (after timeout)

Arne-W, 2020-05-16:
I will try to reproduce this issue tomorrow - but I think a timeout of 2 seconds and a retry of 5 should be enough. You are using macOS 10.15.4 aren't you? I am pretty sure this problem relies on the macOS version - 10.15.4 is known to have a lot of network problems.
If you have a Mac which has an older macOS version - at least 10.15.3 - it would be cool if you could try to reproduce the problem. If not we should wait until the next macOS release.

Arne-W, 2020-05-17:
I was not able to reproduce this issue with 32 waypoints. Writing and fetching works without any problems. It seems the problem is related to macOS 10.15.4.

Arne-W, 2020-06-06:
@muramura we have a new release candidate for macOS. Perhaps you want to test it :smiley: 
You can find it here: [https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1)

billbonney, 2020-06-06:
> 10.15.4 is known to have a lot of network problems
I haven’t experienced any network issues on macOS. Or is this related to Qt 5.9.3?

Arne-W, 2020-06-07:
I work for a company which produces cameras for industrial purpose. Among several special frame grabber interfaces we produce GigE and USB cameras.
Since the update to 10.15.4 a lot of customers have heavy issues when using GigE cameras especially with timeout handling and packet loss. We still hope that those issues are fixed with the next update.