ArduPilot/apm_planner #1212: Ardupolit not working with mac

DroneRookie, 2020-06-15:
Hello Experts,
I recently got a new drone which has ARDUPOLIT, I am not able to connect it with mission planner/APM2  in mac. I couldn't install APM in mac as it said version is not compatible with last Mac OS. Hence I then connect it with windows via virtual machine (parallel). But I am still not able to connect to mission planner.
I am not sure whether my ARDUPOLIT is not working or I am missing any part.
PLease help!!

billbonney, 2020-06-15:
see https://github.com/ArduPilot/apm_planner/releases for latest release that has been tested on macOS Catalina 10.15.5

closed 2020-06-15 by billbonney

DroneRookie, 2020-06-16:
Hello Billbonney, 
Thanks for your reply but as you may have guessed i am new to all this, from above link how do i install ardupilot in mac.when i tried it gave me same error again "not compatible with macOS Catalina.
Any help will be greatly appreciated.

billbonney, 2020-06-16:
if you CTRL-CLICK and then select OPEN it should open. The only error message should be 'unidentified developer' as it's not a 'signed package'

https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac

if that's not the case, can you post a screen grab of the error message. It is working for me and I tested on a clean machine.

DroneRookie, 2020-06-16:
I am already doing that , i am aware of CTRL-CLICK functionality but i am still getting below error.
I have attached all the relevant SS while installing APM planner.

[image]
[image]
[image]

[image]

billbonney, 2020-06-16:
You don’t need the FTDI serial port driver anymore. It’s built into macOS.

DroneRookie, 2020-06-16:
Thanks for your reply.
I didn't exclusively download FTDI serial-orr drivers, its included in the APM installer.
Can you please tell me how do i go about it then?

billbonney, 2020-06-16:
Just drag ‘APM Planner 2.0’ to the Applications folder on your machine and Open it

DroneRookie, 2020-06-16:
Thanks Billbonney,
I could install APM now ( one step done:))
But now when i plugin  my ARDUPOLIT to the laptop and try to install firmware i am getting below error "No valid device found.Chk to be sure your APM2.5+ or  PIXHAWK/PX4 is plugged in and drivers installed"
[image]

mboland, 2020-06-16:
I had similar issues some time ago (years) and did an uninstall and haven’t had issues since.
From the FTDI website you can find their driver manuals which are helpful:

4.2 Uninstalling FTDI VCP Drivers on OS X 10.9 onwards
To remove the drivers from Mac OS X 10.9 onwards, the user must be logged on as root. Root is a reserved username that has the privileges required to access all files.
Start a Terminal session (Go > Applications > Utilities > Terminal) and enter the following commands at the command prompt:
cd /Library/Extensions
rm -r FTDIUSBSerialDriver.kext
then reboot
To temporarily operate as the root user you can use sudo at the beginning of the command e.g.
cd /Library/Extensions
sudo rm -r FTDIUSBSerialDriver.kext
then reboot
The driver will then be removed from the system.

4.3 Uninstalling D2XX Drivers
To uninstall the D2XX driver, simply delete the library and the symbolic link: Start a Terminal session (Go > Applications > Utilities > Terminal)
Change directory to the /usr/local/lib (cd /usr/local/lib)
Delete the library (rm libftd2xx.1.2.2.dylib)
Delete the symbolic link (rm libftd2xx.dylib)
If you have installed D2xxHelper (required for OS X 10.11 onwards), it also must be removed:
sudo rm –r /Library/Extensions/D2xxHelper.kext
Complete the uninstallation by rebooting.

Hope this helps.

DroneRookie, 2020-06-16:
Hello Mboland,
Thanks for your reply.
I have uninstalled above drivers as you had said but still no luck. I am still getting error "No valid device found"
Do i need to install the drivers back from their website or as Billbonney said, new version has these drivers inbuilt.

mboland, 2020-06-16:
Install the NEW version from their website.
Also don’t rely on the USB cables you are using.

DroneRookie, 2020-06-17:
Sorry still not working, installed new drivers and changes USB cable as well.

DroneRookie, 2020-06-18:
Guys, one more step done, one more step conquered.
Now i am able to connect to APM planner and its able to find port as well.
I  tried to install firmware for quadcopter, it got completed saying "APM flashing complete".
Unfortunately that's as far as i could go. Nothing happens after that.
Am I missing anything after that , do i need to do anything else to start my configuration.
I clicked on connect and screen is shown connected but there is no movement on the left side panel.

Arne-W, 2020-06-20:
You should use the [support forum](https://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0/72)... however in your home directory the should be a folder named apmplanner2. Within that folder you should find logfiles. Take a look - perhaps they can help to find out whats going wrong.