ArduPilot/apm_planner #1213: APM Planner not working on MAC OS  Catalina 10.15.3

liox1985, 2020-06-17:
Dear Experts,

I am facing issues to launch APM planner on MAC not sure which version should be used. kindly advice 

kindly help with solution on -nileshdeals@gmail.com 
[image]
[image]
[image]

liox1985, 2020-06-17:
Dear Experts please help

billbonney, 2020-06-17:
From your screenshots it’s working.. Docs here https://ardupilot.org/planner2/index.html 

(Just ignore any instructions to install serial port drivers for FTDI etc..)

liox1985, 2020-06-17:
No its not working there is no option to detect FMU and select the frame to build ..

Install framwork screen is blank

billbonney, 2020-06-17:
support forum is here for using the app https://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0/72

You just need to select your device to see the screen https://ardupilot.org/planner2/docs/connect-autopilot.html

closed 2020-06-17 by billbonney

liox1985, 2020-06-17:
Finally managed to connect it however 
Cannot see preloaded Models and Menu like (https://ardupilot.org/planner2/_images/ap2_connect-autopilot-connected-firmware-
[image]

warning.png) 

How to add them any reference

liox1985, 2020-06-18:
Dear Experts, 

Kindly assist with your inputs to proceed further ..

Arne-W, 2020-06-20:
Please use the support forum - and update to the new version [2.0.28-rc1](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1).

pitops, 2021-08-06:
I have the same issue witht v2.0.28-rc1. Any updates?

Arne-W, 2021-08-10:
@pitops Unfortunately we have no macOS package at the moment - I do not own a mac and cannot build for it. The Bug you are seeing is fixed in the current master - so sorry!

pitops, 2021-08-12:
@Arne-W how easy would it be for me to build it?

Arne-W, 2021-09-01:
@pitops I don't know as I have never build for OSX. But there should be some scripts which should help you. Perhaps @billbonney can help you he has build APM for OSX in the past.,or @AndreiSoroka can help you too he has updated the build info for OSX.

AndreiSoroka, 2021-09-02:
Hi. 
**Try it**: [apmplanner2.zip](https://github.com/ArduPilot/apm_planner/files/7100146/apmplanner2.zip)
last commit in the build: bc6721a6b875ede1116a2a4cb6f9d3d74cd9a6ef
It is my build for MacOs

_____

Yes. I had that problem too: https://github.com/ArduPilot/apm_planner/issues/1213#issuecomment-645292508

I tried other versions 2.x.x and had the same problem.  (On release 2.0.29-rc1 should be fixed?)
So I was tired and decided to build the app from the master branch. And all worked :)

I updated README: https://github.com/ArduPilot/apm_planner#mac-os-x

```bash
# init project and install qt5
brew link qt5 --force
git clone git@github.com:ArduPilot/apm_planner.git;
cd apm_planner;
# build
qmake -spec macx-clang apm_planner.pro;
make -j8;
# open app
open ./release/apmplanner2.app
```

mboland, 2021-09-02:
That zip file is incomplete @AndreiSoroka 
It is only 33.9Mb as apposed to the Sep20202 version at 124.4Mb
[image]
Mac OSX Monterey
[image]

AndreiSoroka, 2021-09-02:
Interesting...
I use Big Sur, but I think it is problem not OS. It is a build problem.
[image]

@Arne-W how to build for production? (I mean, how create a new release build? Maybe need to use something other than modules or plugins or builders or flags or something else? )

@mboland do you tried to build on your system?

P.s. I'm not a c++ developer :)

mboland, 2021-09-02:
I’m no C++ dev either, although given the recipe and dependancies I could muddle through

AndreiSoroka, 2021-09-02:
@mboland 
can you try?
```bash
brew link qt5 --force
git clone git@github.com:ArduPilot/apm_planner.git;
cd apm_planner;
qmake -spec macx-clang apm_planner.pro;
make -j8;
open ./release/apmplanner2.app
```
I think should be work

mboland, 2021-09-02:
Cant get past the first line:

michaelboland@Mac-Pro ~ % brew link qt5 --force

Traceback (most recent call last):
	11: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `<main>'
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `require_relative'
	 9: from /usr/local/Homebrew/Library/Homebrew/global.rb:26:in `<top (required)>'
	 8: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 7: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:1:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:19:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:52:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:23:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:369:in `initialize': Version value must be a string; got a NilClass () (TypeError)

AndreiSoroka, 2021-09-02:
@mboland 
 brew -v?
```bash
andreisoroka@Andreis-MacBook-Pro-2 release % brew -v
Homebrew 3.2.6
Homebrew/homebrew-core (git revision 014a823eead; last commit 2021-08-10)
Homebrew/homebrew-cask (git revision 5e2bced8b6; last commit 2021-08-10)
```

mboland, 2021-09-02:
michaelboland@Mac-Pro ~ % brew -v
Traceback (most recent call last):
	11: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `<main>'
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `require_relative'
	 9: from /usr/local/Homebrew/Library/Homebrew/global.rb:26:in `<top (required)>'
	 8: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 7: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:1:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:19:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:52:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:23:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:369:in `initialize': Version value must be a string; got a NilClass () (TypeError)

AndreiSoroka, 2021-09-02:
You have a problem with brew.  I don't know is brew supporting OS version Montery?

```bash
xcode-select --install;
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh);
# afret that
brew -v
```

from https://brew.sh/ and https://docs.brew.sh/Installation (+ info: how uninstall brew)

mboland, 2021-09-02:
Command line tools weren’t installed (Apple do that every upgrade !!)
Installing and will try again.

mboland, 2021-09-02:
The installation of Xcode CL tools didn’t make any difference
Dont have the time this morning to be mucking around with this, and this is where building leaves me cold.

***@***.*** ~ % brew -v
Traceback (most recent call last):
	11: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `<main>'
	10: from /usr/local/Homebrew/Library/Homebrew/brew.rb:17:in `require_relative'
	 9: from /usr/local/Homebrew/Library/Homebrew/global.rb:26:in `<top (required)>'
	 8: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 7: from /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	 6: from /usr/local/Homebrew/Library/Homebrew/os.rb:1:in `<top (required)>'
	 5: from /usr/local/Homebrew/Library/Homebrew/os.rb:19:in `<module:OS>'
	 4: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:52:in `prerelease?'
	 3: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `version'
	 2: from /usr/local/Homebrew/Library/Homebrew/os/mac.rb:18:in `new'
	 1: from /usr/local/Homebrew/Library/Homebrew/os/mac/version.rb:23:in `initialize'
/usr/local/Homebrew/Library/Homebrew/version.rb:369:in `initialize': Version value must be a string; got a NilClass () (TypeError)

Arne-W, 2021-09-03:
The problem is that the zip file does not contain the dependencies (Qt libraries) therefore it is much smaller. @AndreiSoroka take a look at this script [https://github.com/ArduPilot/apm_planner/blob/master/scripts/deploy_osx.sh](https://github.com/ArduPilot/apm_planner/blob/master/scripts/deploy_osx.sh). It shows how the older versions were packed for deployment. The Script will not work as the Qt version is hardcoded. But creating an dmg package should work this way. I would do it step by step in a terminal. The key to success are the usage of the macdeployqt and the hditool. 
Hope this helps!

AndreiSoroka, 2021-09-03:
@Arne-W what is apm_planner2_target.sparseimage?

Arne-W, 2021-09-09:
@AndreiSoroka Thank you very much for your effort!
Did anyone test the package Andrei built?

billbonney, 2021-09-09:
> apm_planner2_target.sparseimage?

It’s the way to dtribute the app with a readme and other files. You don’t really need it anymore.

JonAllen816, 2022-02-06:
Downloaded and run the package Andrei built on Monterey 12.2 and I get the following error upon opening apmplanner2:

Unable to load ApmToolbar.qml. Please reinstall the application and try again. Errors are as follows:
file:///Applications/apmplanner2.app/Contents/MacOS/qml/ApmToolBar.qml:17:1: plugin cannot be loaded for module "QtQuick": Cannot load library /Applications/apmplanner2.app/Contents/MacOS/qml/QtQuick.2/libqtquick2plugin.dylib: (dlopen(/Applications/apmplanner2.app/Contents/MacOS/qml/QtQuick.2/libqtquick2plugin.dylib, 0x0085): Library not loaded: /usr/local/Cellar/qt@5/5.15.2/lib/QtQuick.framework/Versions/5/QtQuick
  Referenced from: /Applications/apmplanner2.app/Contents/MacOS/qml/QtQuick.2/libqtquick2plugin.dylib
  Reason: tried: '/usr/local/Cellar/qt@5/5.15.2/lib/QtQuick.framework/Versions/5/QtQuick' (no such file), '/Library/Frameworks/QtQuick.framework/Versions/5/QtQuick' (no such file), '/System/Library/Frameworks/QtQuick.framework/Versions/5/QtQuick' (no such file))