ArduPilot/apm_planner #1214: Not able to view the Models and Menu after opening Initial setup (PX4 Connected)

liox1985, 2020-06-18:
Dear Experts,

Not able to view the menu after connecting PX4 (FMU2.x) using USB port 
Apple OS - 10.15.3 (Catalina)
APM Version 2.0.26-2c1 

Kindly assist .. Please refer attached pic

[image]

liox1985, 2020-06-18:
[image]

Arne-W, 2020-06-20:
Please stop spamming

closed 2020-06-20 by Arne-W

liox1985, 2020-06-23:
Hi Team , Its not spam facing genuine issue with software to connect.

Please advice.

billbonney, 2020-06-23:
Use the latest release https://github.com/ArduPilot/apm_planner/releases and please look on the forum. (You have askes this question twice hence the spam reference)