ArduPilot/apm_planner #1215: Undefined reference errors in libQt5Quick.so, libQt5Qml.so

Eric97, 2020-06-19:
Hello community,

I am testing the `apm_planner` master branch with Qt 5.95 on Ubuntu 18.04. During the linking state, came across the following undefined errors: 

> build-release/obj/AP2DataPlotThread.o: In function `_GLOBAL__sub_I_AP2DataPlotThread.cc':
AP2DataPlotThread.cc:(.text.startup+0x184): undefined reference to `QColor::setNamedColor(QLatin1String)'
build-release/obj/main.o: In function `_GLOBAL__sub_I_main.cc':
main.cc:(.text.startup+0x1d8): undefined reference to `QColor::setNamedColor(QLatin1String)'
build-release/obj/QGCCore.o: In function `_GLOBAL__sub_I_QGCCore.cc':
QGCCore.cc:(.text.startup+0x1d8): undefined reference to `QColor::setNamedColor(QLatin1String)'
build-release/obj/UASManager.o: In function `_GLOBAL__sub_I_UASManager.cc':
UASManager.cc:(.text.startup+0x174): undefined reference to `QColor::setNamedColor(QLatin1String)'
build-release/obj/UAS.o: In function `_GLOBAL__sub_I_UAS.cc':
UAS.cc:(.text.startup+0x186): undefined reference to `QColor::setNamedColor(QLatin1String)'
build-release/obj/LinkManager.o:LinkManager.cc:(.text.startup+0x184): more undefined references to `QColor::setNamedColor(QLatin1String)' follow
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QTouchEvent::TouchPoint::rotation() const@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QTouchEvent::TouchPoint::ellipseDiameters() const@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::isCommonTextEditShortcut(QKeyEvent const*)@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::QInputControl(QInputControl::Type, QObject*)@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QOpenGLShaderProgram::addCacheableShaderFromSourceCode(QFlags<QOpenGLShader::ShaderTypeBit>, char const*)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QT_DISTANCEFIELD_SCALE(bool)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::staticMetaObject@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::qt_metacall(QMetaObject::Call, int, void**)@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QPointingDeviceUniqueId::staticMetaObject@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QOpenGLShaderProgram::addCacheableShaderFromSourceCode(QFlags<QOpenGLShader::ShaderTypeBit>, QByteArray const&)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QT_DISTANCEFIELD_BASEFONTSIZE(bool)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Svg.so: undefined reference to `qt_version_tag@Qt_5.9'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QTouchEvent::TouchPoint::uniqueId() const@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::qt_metacast(char const*)@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `typeinfo for QInputControl@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QT_DISTANCEFIELD_HIGHGLYPHCOUNT()@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::QInputControl(QInputControl::Type, QObjectPrivate&, QObject*)@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `vtable for QInputControl@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Qml.so: undefined reference to `QDateTime::QDateTime(QDateTime&&)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QPointingDeviceUniqueId::fromNumericId(long long)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QColor::setNamedColor(QLatin1String)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QOpenGLShaderProgram::addCacheableShaderFromSourceCode(QFlags<QOpenGLShader::ShaderTypeBit>, QString const&)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QInputControl::isAcceptableInput(QKeyEvent const*) const@Qt_5_PRIVATE_API'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QT_DISTANCEFIELD_RADIUS(bool)@Qt_5'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libQt5Quick.so: undefined reference to `QStringRef::isRightToLeft() const@Qt_5'
collect2: error: ld returned 1 exit status
Makefile:1797: recipe for target 'release/apmplanner2' failed

Is it due to version incompatibility in Qt? But think 5.9.5 is officially supported based on readme. 

Regards,
Eric

Arne-W, 2020-06-20:
You could use the official released AppImage you can find [here](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1).
did you install "qtdeclarative5-qtquick2-plugin"?

Eric97, 2020-06-21:
Thanks.  `qtdeclarative5-qtquick2-plugin` is installed. I tried a fresh build on a clean Ubuntu 18.04, then compiles without problems. Guess there is a conflict with my other Qt dependent packages.

closed 2020-06-21 by Eric97