ArduPilot/apm_planner #1216: Error message prearm: compass not healthy

DroneRookie, 2020-06-20:
Hello Experts,
Can any one help me with this ? I am new to drone world so excuse me for silly question.
I am getting an error message when i am trying to configure APM planner "PreArm: Compass not healthy " and don’t know how to fix it and its not letting me complete the configurations.
Also, i could configure RC in the APM planner  and see that RC is connected to drone but its not making arming the motors( which was happening earlier, before i configured RC)
Any help will be highly appreciated.

Arne-W, 2020-06-20:
I think you will have to calibrate the compass.
Please go to the [support forum](https://discuss.ardupilot.org/) there you will find a lot of informations regarding your problems.
This is only for software problems related to apmplanner itself. 
Thank you

DroneRookie, 2020-06-20:
Thanks for your Arne-W.
I have tried to calibrate compass numerous times but i keep on getting error that "not enough data point" and calibration not complete.I have moved APM in all directions in order to get enough data points but i keep getting the same message.
Anyways thx for your replay

Arne-W, 2020-06-20:
Here you can find some hints: [https://youtu.be/DmsueBS0J3E](url)
And stay away from magnetic things or metal when doing calibration.

closed 2020-06-20 by Arne-W